Open-Agriculture/AgIsoStack-plus-plus #41: Define a CMake style for the repo

RFRIEDM-Trimble, 2022-11-10:
For the 1.2% of this codebase that is cmake, there should be a coding standard that can be linted against in PR's. This ensures contributors are abiding by a common set of guidelines, just like the C++ code. 

**Proposal**
Use [cmake-lint](https://cmake-format.readthedocs.io/en/latest/cmake-lint.html) and [cmake-format](https://cmake-format.readthedocs.io/en/latest/cmake-format.html)

**Implementation**
Once the tooling and styles are decided, a PR can be issued to do the following:
1. Add the necessary configuration files if defaults are not used
1. Determine how the linter/formatter can know which files need to be formatted/linted if automatic recursive detection is not supported
1. Add information to the README on the usage of the linter and formatter
1. Go through and fix all current lint/format issues in the codebase
1. Future: Add support to call the linter and/or formatter from cmake as part of CI to block non-compliant PR's

Thoughts?

ad3154, 2022-11-10:
It's probably a good idea, similar to how we have a clang-format file for linting the C++ files. In a broader sense that process needs additional documentation and CI checking as well. Might make sense to put those in a CONTRIBUTING.md file.

For CI we might be able to make a GitHub action that runs the linters and then does a `git diff` to ensure no lines were changed. If no lines were changed, then the files are ready for PR. If any line changed, then block the PR. Or perhaps there's a pre-baked one out there somewhere, I'd have to do some research.

ad3154, 2022-12-27:
Haven't forgotten about this. I do have one open issue on cmake-lint here: https://github.com/cmake-lint/cmake-lint/issues/17

But when we run cmake-format, it comes back pretty clean with mostly line length complaints so it should be pretty easy to get this completed and closed out.

RileyKenyon, 2023-01-01:
Hey @ad3154 - I stumbled upon this thread and wanted to check in on your cmake-lint issue.

The `cmake-lint` script belongs to the [cmakelang](https://github.com/cheshirekow/cmake_format/blob/master/cmakelang/format/__main__.py#L314) module: 

From those docs, it looks like the format has to be yaml/json/python.
```
cmake-lint [-h]
           [--dump-config {yaml,json,python} | -o OUTFILE_PATH]
           [-c CONFIG_FILE]
           infilepath [infilepath ...]
```

The other module [cmakelint](https://github.com/cmake-lint/cmake-lint) where you filed your issue has different syntax:
```
cmakelint -h

Syntax: cmakelint.py [--version] [--config=file] [--filter=-x,+y] [--spaces=N]
                     [--quiet] [--linelength=digits]
        <file> [file] ...
```
If you were to run your config with `cmakelint` instead:
```
cmakelint --config=.cmakelintrc CMakeLists.txt 
Total Errors: 0
```

But if you try to do:
```
cmake-lint --config=.cmakelintrc CMakeLists.txt
...
6, in try_get_configdict
    raise RuntimeError("Failed to parse {} as any of yaml, json, or python"
RuntimeError: Failed to parse .cmakelintrc as any of yaml, json, or python
```

If you want to run linting with the `cmakelang` instead:
```
cmake-lint --dump-config 'json' > config.json
cmake-lint --config=config.json CMakeLists.txt
```
Hope that helps!

PR #148 (2023-01-23): [#41]: Add cmake-lint and cmake-format checks to GitHub actions
**What's new:**
- Added the cmake-format configuration file: `.cmake-format.yaml`.
- Combined clang-format and cmake-format checks into one GitHub action with multiple jobs.
- Ran cmake-format on the current CMakeLists.txt files, such that the cmake-format and cmake-lint accepts them.
- Set `find_package(Isobus REQUIRED)` if the `BUILD_EXAMPLES` is not set, instead of always checking `CONFIG`.
- Added cmake-format command to format all `CMakeLists.txt` files to `CONTRIBUTING.md`.

Closes #41
closed 2023-01-23 by ad3154