Open-Agriculture/AgIsoStack-plus-plus #57: Implement support for find_package in CONFIG mode

RFRIEDM-Trimble, 2022-11-15:
Please add support for finding this package with a call to `find_package` in CONFIG mode. 

It looks like `CMakePackageConfigHelpers` is included, but not used to create the package configuration file or version file. 
[Reference](https://cmake.org/cmake/help/latest/module/CMakePackageConfigHelpers.html): 

The utility of this is you can install the library in your system headers or somewhere on your CMAKE_PREFIX_PATH, then just call find_package in the consuming application. 

Note: There is a guide [here](https://cmake.org/cmake/help/book/mastering-cmake/chapter/Finding%20Packages.html#creating-cmake-package-configuration-files), but that is the manual way of doing it, CMakePackageConfigHelpers alleviates a lot of the boilerplate.

ad3154, 2022-11-15:
Hmm, if I add this to CMakeLists.txt, it generates and installs a package version file, but for some reason CMake doesn't like it...
```
write_basic_package_version_file( ${CMAKE_CURRENT_SOURCE_DIR}/cmake/isobusConfig.cmake VERSION ${PROJECT_VERSION} 
COMPATIBILITY SameMajorVersion)

install(FILES
${CMAKE_CURRENT_SOURCE_DIR}/cmake/isobusConfig.cmake
DESTINATION ${CMAKE_INSTALL_LIBDIR}/cmake/${PROJECT_NAME})
```

From another project that does `find_package(isobus 0.1 REQUIRED)`

```
CMake Error at CMakeLists.txt:13 (find_package):
  Could not find a configuration file for package "isobus" that is compatible
  with requested version "0.1".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/isobus/isobusConfig.cmake, version: unknown

-- Configuring incomplete, errors occurred!
```

Any thoughts on what I might be missing?

RFRIEDM-Trimble, 2022-11-15:
Push it on a branch and I'll try it?

ad3154, 2022-11-15:
> Push it on a branch and I'll try it?

Branch is [meta/cmake-package-version-file](https://github.com/ad3154/ISO11783-CAN-Stack/tree/meta/cmake-package-version-file)

RFRIEDM-Trimble, 2022-11-15:
You misnamed the [version file.](https://cmake.org/cmake/help/latest/manual/cmake-packages.7.html#package-version-file)

It needs to be isobus-config-version.cmake or isobusConfigVersion.cmake. Also, setting version to PROJECT_VERSION is redundant, that's already the default. 

Once I changed that, the package can be found, but the targets aren't part of it (yet)

RFRIEDM-Trimble, 2022-11-15:
I would follow this guide; https://cliutils.gitlab.io/modern-cmake/chapters/install/installing.html

Working on a fix, can push it to another branch for your review.

ad3154, 2022-11-15:
> I would follow this guide; https://cliutils.gitlab.io/modern-cmake/chapters/install/installing.html
> 
> Working on a fix, can push it to another branch for your review.

I appreciate the help & resources, this is a bit outside my CMake knowledge so it's a learning process for me :)

RFRIEDM-Trimble, 2022-11-15:
Check out commit bece837a7165423776f5faaeaa13790ba6659b3a

That does get the export working, but it uncovered the include directories are not portable. 

Yea, I have to look up the tutorials every time, the cmake docs themselves have you using finicky functions and never recommend not using something where there are many ways to do it. 

```
CMake Error in CMakeLists.txt:
  Imported target "isobus::Isobus" includes non-existent path

    "ISO11783-CAN-Stack/install/ISOBUS/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
  ```

RFRIEDM-Trimble, 2022-11-15:
Ah, it's because this line in isobus/CMakelists.txt:
```
target_include_directories(Isobus PUBLIC
  $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
  $<INSTALL_INTERFACE:${ISOBUS_PUBLIC_NAME}/include>
 ) 
 ```
 
This conflicts with 
```
install( TARGETS Isobus SystemTiming HardwareIntegration
         EXPORT isobusTargets
         LIBRARY DESTINATION lib
         ARCHIVE DESTINATION lib
         RUNTIME DESTINATION bin
         INCLUDES DESTINATION include
)
```

Per `INCLUDES DESTINATION` docs: 

```
This option specifies a list of directories which will be added to the [INTERFACE_INCLUDE_DIRECTORIES](https://cmake.org/cmake/help/latest/prop_tgt/INTERFACE_INCLUDE_DIRECTORIES.html#prop_tgt:INTERFACE_INCLUDE_DIRECTORIES) target property of the <targets> when exported by the [install(EXPORT)](https://cmake.org/cmake/help/latest/command/install.html?highlight=install#install-export) command. If a relative path is specified, it is treated as relative to the $<INSTALL_PREFIX>.
```

Basically, this doesn't actually install the header files anywhere to the install directory. They need to be added. 
The modern-cmake cpaters installing resource also has a not on this:
```
The various destinations are only needed if you have a library, static library, or program to install. The includes destination is special; since a target does not install includes. It only sets the includes destination on the exported target (which is often already set by target_include_directories, so check the MyLibTargets file and make sure you don't have the include directory included twice if you want clean cmake files).
```

The library is definitely doing it twice right now. I have to do some research to figure out what's better. 
 
 Do you want the public name to be `ISOBUS` or `isobus`?

RFRIEDM-Trimble, 2022-11-16:
Also, do you want the includes flat, or in folders? IE, do you want consumers typing 
`#include <can_hardware_interface.hpp>`
OR
`#include <hardware_integration/can_hardware_interface.hpp>`
OR
`#include <ISOBUS/hardware_integration/can_hardware_interface.hpp>`

I'd recommend against the first one. For example, from boost: `#include <boost/lambda/lambda.hpp>`

ad3154, 2022-11-16:
I think my preference is `isobus` as the public name, as well as being able to do `#include <isobus/hardware_integration/can_hardware_interface.hpp>` rather than the flat structure, as I agree it leaves out a lot of implicit context for consumers

RFRIEDM-Trimble, 2022-11-16:
Good. This leaves two approaches. 

1. Use `install(DIRECTORY)` with a  file matching `.hpp` pattern. I think this would mean moving install to top level only. 
2. Upgrade cmake version to build the library to 3.23 to support file sets, which actually solves this issue the right way. This does mean consumers who want to submodule or compile from source will need to upgrade. For consumers who don't want to upgrade cmake, they can consume the debian or prebuilt binaries at some release cadence.

ad3154, 2022-11-16:
I'm hesitant to jump CMake versions past 3.16 (the default for Ubuntu 20.04, raspian, and maybe RHEL) also, so I am thinking option number 1 might be the way to go!

RFRIEDM-Trimble, 2022-11-16:
For approach 1, I got it working here: 6a0cea3f20508967f1061c7f853821f873dbbc21

It's a bit of hack because the installed header files use the flat include style, so you have to have targets support both. Overall, it seems to work fine, so let me know if you agree with the behavior there. 

Important bits are as follows:

Consumer example include:
```
#include "isobus/isobus/can_general_parameter_group_numbers.hpp"
#include "isobus/hardware_integration/can_hardware_interface.hpp"
#include "isobus/isobus/can_network_manager.hpp"
#include "isobus/isobus/isobus_diagnostic_protocol.hpp"
#include "isobus/hardware_integration/socket_can_interface.hpp"
```

The install tree looks like this:
```
$ tree install/
install/
├── include
│   └── isobus
│       ├── hardware_integration
│       │   ├── can_hardware_interface.hpp
│       │   ├── can_hardware_plugin.hpp
│       │   └── socket_can_interface.hpp
│       ├── isobus
│       │   ├── can_address_claim_state_machine.hpp
│       │   ├── can_badge.hpp
│       │   ├── can_callbacks.hpp
│       │   ├── can_constants.hpp
│       │   ├── can_control_function.hpp
│       │   ├── can_extended_transport_protocol.hpp
│       │   ├── can_frame.hpp
│       │   ├── can_general_parameter_group_numbers.hpp
│       │   ├── can_hardware_abstraction.hpp
│       │   ├── can_identifier.hpp
│       │   ├── can_internal_control_function.hpp
│       │   ├── can_managed_message.hpp
│       │   ├── can_message.hpp
│       │   ├── can_NAME_filter.hpp
│       │   ├── can_NAME.hpp
│       │   ├── can_network_configuration.hpp
│       │   ├── can_network_manager.hpp
│       │   ├── can_parameter_group_number_request_protocol.hpp
│       │   ├── can_partnered_control_function.hpp
│       │   ├── can_protocol.hpp
│       │   ├── can_transport_protocol.hpp
│       │   ├── can_warning_logger.hpp
│       │   ├── isobus_diagnostic_protocol.hpp
│       │   └── isobus_virtual_terminal_client.hpp
│       └── utility
│           ├── iop_file_interface.hpp
│           ├── processing_flags.hpp
│           ├── system_timing.hpp
│           └── to_string.hpp
└── lib
    ├── cmake
    │   └── isobus
    │       ├── isobusConfig.cmake
    │       ├── isobusConfigVersion.cmake
    │       ├── isobusTargets.cmake
    │       └── isobusTargets-debug.cmake
    ├── libHardwareIntegration.so
    ├── libIsobus.so
    └── libSystemTiming.so

8 directories, 38 files
```
You can see the properly nested includes by folder. You also get the cmake libs (.so) in lib/, and ALL of the cmake scripts get dumped into lib/cmake/isobus. 

I think that approach is fine, just be aware if this library gets really big, you start supporting COMPONENTS level install and finding to modularize it, perhaps from optional GUI things, this install method will install all the header files regardless if the library was actually built. It's not too much work to change in the future. File sets support this, by only installing headers for something that was actually built and requested to install.

ad3154, 2022-11-20:
Looks OK to me - if you put together that PR I would be happy to review and approve it.

RFRIEDM-Trimble, 2022-11-21:
Ok. Looks like I have to update all the internal include paths to be the same, otherwise the examples can't build from top level. Shouldn't take too long.

ad3154, 2022-12-03:
@RFRIEDM-Trimble  Now that #58 is merged is this issue closable?

RFRIEDM-Trimble, 2022-12-03:
Yes, it can be closed. Thanks for the patience as I worked though all the mods.
This should greatly increase the usability of this package to be integrated into higher level applications.

closed 2022-12-03 by RFRIEDM-Trimble