Open-Agriculture/AgIsoStack-plus-plus #65: VT example should support different widths and heights of data masks

GwnDaan, 2022-12-09:
The terminal on which the example was tested supports 2 pixel masks modes. One has a 260x260 pixel mask and the other is 600x600 pixels. 

The example gets clipped when running on the 260x260mask:
[image]

While the 600x600 mask displays:
[image]

Therefore I'm assuming the VT example implements a data mask of around 500x500 pixels. What would be the best approach to support different masks? Is there anything written about this in the iso 11783 standard?

tomstix, 2022-12-26:
As far is I know, a common practice is to just use 2-3 different Object Pools for common resolutions. But automatic scaling would surely be a very useful feature.

GwnDaan, 2022-12-28:
I think https://github.com/moehman/PoolEditParser has successfully implemented scaling of objects. Haven't dug deep into it but in their readme it states: 
> Because the XML file is parsed at run-time when the VT properties are known, the PoolEdit parser can resize the GUI objects and reduce the number of colors 

They use a XML file format to save all the parameters of the objects, but parsing these parameters from the binary format shouldn't be too hard either I believe. I think the latter is preferred as it doesn't require any new file format, which means that support for binary object pools remains intact.

GwnDaan, 2023-01-12:
Hi, I think the workflow could look something like this? I'm just thinking out loud here and it looks pretty doable. As far as I can see the only hard/boring part would be to specify the length and indexes (for width/height) of each pool? Let me know what you think.

[image]

I did not consider different aspect ratios here because the standard says in section 4.5.2 about data masks: "Higher resolution mask areas are permitted, but the square aspect ratio shall be strictly enforced.". 
Although the dimensions for soft keys masks don't have this requirement. I think we can just apply a "fit" action for now, and implement any other action later if necessary. (E.g. "fill" or "center fit", etc..)

ad3154, 2023-01-12:
Generally this would work if all pools can be stored in RAM, but for RAM constrained platforms it might not work, as they'd be using the callback method to fetch pool data in segments for upload. I think we may be able to just parse the objects as they're being uploaded and apply the multiplier at upload time in those cases. This would be stateful though and relies on the pool being at least mostly error-free. Ideally we'd probably provide both a pre-processing solution and a run-time manipulation solution I suppose? I can work on this issue next potentially - I am working on a VT server application on the side so I have VT stuff on the mind, haha.

GwnDaan, 2023-01-12:
Sounds good! Not sure what a pre-processing solution would look like though in the context of this stack. I'm just frustrated by my test bench not having the same dimensions as the display I'm using in the tractor, especially combined with an object pool creator I cobbled together myself that doesn't really output multi dimension pools easily.

ad3154, 2023-01-13:
Well, similar to how we have that little function to load an IOP file, we could have a little utility function, or a static function on the VT Client class that goes through and scales every object and produces a new, scaled IOP file or something. That'd be easier I think than parsing the pool as a stream while we upload it, just less efficient I suppose.

ad3154, 2023-01-13:
Also you made your own pool creator!? Wow! That sounds like a ton of work.

GwnDaan, 2023-01-13:
>Well, similar to how we have that little function to load an IOP file, we could have a little utility function, or a static function on the VT Client class that goes through and scales every object and produces a new, scaled IOP file or something

I see, though that would mean the object pool is not "automatically" adjusted to the "right" dimensions provided by the UT right? Since it will be independent from the state of the VT client class (which holds the UT dimensions). Don't get me wrong, it can still be a very useful feature and better than nothing.
>Also you made your own pool creator!? Wow! That sounds like a ton of work.

Hahah it's not too bad, nothing fancy. Just a small python based script where I can manually specify each object in the pools and generate an usable IOP file format from it.

ad3154, 2023-01-13:
I started writing some of the auto-scaling code [here](https://github.com/ad3154/ISO11783-CAN-Stack/compare/main...feature/vt-autoscaling) that we could do while uploading, and I actually think it's not as tough as I thought it would be. I basically need to just finish writing a read-ahead buffer in the protocol's data chunk callback, read in an object and its child references, scale + reposition them if needed, then provide that buffer content back to the stack. So yeah, maybe this will be less code than I anticipated... the tricky thing is if someone provides a garbage object pool we could get into some trouble. This is probably OK though, as there's not a ton we can do as a VT client with garbage data anyways, so we kind of have to assume the pool is at least properly structured.

ad3154, 2023-01-13:
Oh, I also might need to add separate scaling for things contained by soft key masks? Their scaling multiplier could be different than the one for a data mask area...

GwnDaan, 2023-01-13:
Looking good! Soft key masks need a separate scaling multiplier indeed. I just thought of another problem where it might be not possible to apply a scaling factor to fit the pool "exactly" to the dimensions of the VT. This because the width/height or positioning x/y after applying the scaling factor could end up in decimal format. I can't get my head around if rounding this number up/down would be a solid fix, or that it will end up with visual artifacts.

ad3154, 2023-01-13:
Yeah, rounding might be an issue, I'm not sure. Right now it would be truncating it after multiplying the scale, so objects might be a tiny bit smaller than "perfectly scaled". I'm not sure how good or bad it will look yet, but we may need to do smarter rounding depending on how it goes. We also need to implement font substitution, which will be fun

ad3154, 2023-01-23:
Progress update: Still a work in progress... getting close to a first draft. Some objects in the Vt example pool are getting scaled on-the-fly now but the autoscaler is still choking on the largest objects, namely output strings with a non 0 value (because for some silly reason, ISO/AEF put the macro count AFTER the entire string unlike other objects 😑 so I can't get the total object size without consuming a potentially very large string in RAM) and picture graphics because they are similarly huge. So, getting close. Just need to wrap up those special cases I think, then tidy it all up into nicer, smaller functions. I'm moving a bit slower on this than I had hoped because I'm having to manually validate that it's working object-by-object. 
[image]

ad3154, 2023-01-23:
Oh, also font re-mapping. Still need to implement that, but that should be pretty easy...

ad3154, 2023-01-30:
Another update, I'm fighting a persistent bug and quite a lot of complexity in the on-the fly resizing. I think I might shift gears a little bit to finish up resizing the object pool in-memory (still dynamically, but requires holding onto a copy of the pool while we resize it) then I'll come back to this more complex approach to try and finish it up. That way at least I can stop delaying other features that might build on this.

PR #164 (2023-02-11): [#65] Add dynamic object pool scaling
## What's New
* The VT Client can now automatically scale an object pool to your destination VT! 🥳

Here are some screenshots from my VT showing some different scale factors. My terminal has 480x480 data mask area, and 72x72 soft key designators.

[image]

Closes #65 

### Minor stuff
* Made one minor update to the example object pool to fix some object alignment.
* Cleaned up some lingering doxygen warnings.
* Clarified tutorial build instruction
closed 2023-02-11 by ad3154