Open-Agriculture/AgIsoStack-plus-plus #70: Add documentation of CANStackLogger usage

GwnDaan, 2022-12-12:
I couldn't find any concrete implementation of CANStackLogger in the tutorials/examples. We could consider adding it to the existing examples. Displaying the error/warning messages by default for the VT example could be important for unexperienced users testing this library.

ad3154, 2022-12-15:
This is a bug stemming from the issue that the function signature for `get_can_stack_logger` is `bool CANStackLogger::get_can_stack_logger(CANStackLogger *canStackLogger)` but should be `bool CANStackLogger::get_can_stack_logger(CANStackLogger *&canStackLogger)` (note the missing & 🙄)

ad3154, 2022-12-15:
Added a fix and documentation to this PR for your review @GwnDaan #74

PR #74 (2022-12-15): [Core/Meta]: Custom logger fix and documentation
## What's new
* Fixes a segfault when using a custom logger
* Adds tutorial and example documentation for implementing a custom logger

Closes #70
closed 2022-12-15 by ad3154