Open-Agriculture/AgIsoStack-plus-plus #81: Add remaining activation messages

GwnDaan, 2022-12-18:
I saw that only the first 4 activation messages were implemented right now (ButtonActivation, SoftKeyActivation, PointingEvent and SelectInputObject). Since there are 11 in total at this time, that means that there are still 7 missing. For the sake of convenience, do we want to implement these as well? I don't mind spending some time on this myself to get this done.

GwnDaan, 2022-12-18:
I'm talking about the callbacks here

ad3154, 2022-12-18:
You are correct that support for some of them are missing, and we definitely need to add the others. If you're interested in working on them, that would be very helpful! Things have been busy on my end recently so I've spent a bit less time on this as the holidays grow near.

ad3154, 2022-12-18:
The numeric value and string ones are probably the most important if I had to guess

GwnDaan, 2022-12-18:
No worries I'll work on this
> The numeric value and string ones are probably the most important if I had to guess
Yup these are exactly the ones I needed on my end

PR #82 (2022-12-20): [VT] Add remaining activation message callbacks
Adds support for the 7 VT activation messages in the form of callbacks which were still missing.

Closes #81