Open-Agriculture/AgIsoStack-plus-plus #102: Examples Seg Fault on SIGINT

RileyKenyon, 2023-01-01:
When running the examples, exiting the program with `ctrl+c` results in a segfault. Seems to be an issue with the isobus::InternalControlFunction `shared_ptr`. May have something to do with the operations allowed for a [signal handler](https://en.cppreference.com/w/cpp/utility/program/signal).

Easy fix is to to move the creation of the TestInternalECU into main(), although I am not sure how to fix the root cause.

Steps to replicate:
```
cmake -S . -B build -DCMAKE_BUILD_TYPE=DEBUG -DBUILD_EXAMPLES=ON
cmake --build build
gdb build/examples/nmea2000/NMEA2KExampleTarget

# Stop GDB from catching the SIGINT
handle SIGINT noprint pass

# Run the program
run

# Hit ctrl+c
Thread 1 "NMEA2KExampleTa" received signal SIGSEGV, Segmentation fault.
0x00007ffff7cbfdde in std::__detail::_List_node_base::_M_unhook() () from /lib/x86_64-linux-gnu/libstdc++.so.6
```

Here is my back trace:
```
#0  0x00007ffff7cbfdde in std::__detail::_List_node_base::_M_unhook()
    () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x000055555556256c in std::__cxx11::list<isobus::InternalControlFunction*, std::allocator<isobus::InternalControlFunction*> >::_M_erase (
    this=0x55555559e360 <isobus::InternalControlFunction::internalControlFunctionList[abi:cxx11]>, __position=0x5555555b1970)
    at /usr/include/c++/11/bits/stl_list.h:1922
#2  0x00005555555622c9 in std::__cxx11::list<isobus::InternalControlFunction*, std::allocator<isobus::InternalControlFunction*> >::erase (
    this=0x55555559e360 <isobus::InternalControlFunction::internalControlFunctionList[abi:cxx11]>, __position=0x5555555b1970)
    at /usr/include/c++/11/bits/list.tcc:158
#3  0x0000555555561da3 in isobus::InternalControlFunction::~InternalControlFunction (this=0x5555555b1970, __in_chrg=<optimized out>)
    at /home/rkenyon/repos/ISO11783-CAN-Stack/isobus/src/can_internal_control_function.cpp:33
#4  0x00005555555597bf in __gnu_cxx::new_allocator<isobus::InternalControlFunction>::destroy<isobus::InternalControlFunction> (
    this=0x5555555b1970, __p=0x5555555b1970)
    at /usr/include/c++/11/ext/new_allocator.h:168
#5  0x0000555555559783 in std::allocator_traits<std::allocator<isobus::InternalControlFunction> >::destroy<isobus::InternalControlFunction> (
    __a=..., __p=0x5555555b1970)
    at /usr/include/c++/11/bits/alloc_traits.h:535
--Type <RET> for more, q to quit, c to continue without paging--
#6  0x000055555555963f in std::_Sp_counted_ptr_inplace<isobus::InternalControlFunction, std::allocator<isobus::InternalControlFunction>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x5555555b1960) at /usr/include/c++/11/bits/shared_ptr_base.h:528
#7  0x00005555555585eb in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555555b1960) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#8  0x00005555555581af in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x55555559c208 <TestInternalECU+8>, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#9  0x000055555555810e in std::__shared_ptr<isobus::InternalControlFunction, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x55555559c200 <TestInternalECU>, 
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#10 0x000055555555812e in std::shared_ptr<isobus::InternalControlFunction>::~shared_ptr (this=0x55555559c200 <TestInternalECU>, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr.h:122
#11 0x00007ffff7845495 in __run_exit_handlers (status=2, listp=0x7ffff7a19838 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true)
    at ./stdlib/exit.c:113
#12 0x00007ffff7845610 in __GI_exit (status=<optimized out>) at ./stdlib/exit.c:143
#13 0x0000555555557b50 in signal_handler (signum=2) at /home/rkenyon/repos/ISO11783-CAN-Stack/examples/nmea2000/main.cpp:50
#14 <signal handler called>
#15 0x00007ffff78e5868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffffffdbe0, rem=0x7fffffffdbe0)
    at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#16 0x00007ffff78ea6e7 in __GI___nanosleep (req=<optimized out>, rem=<optimized out>) at ../sysdeps/unix/sysv/linux/nanosleep.c:25
#17 0x00005555555583ec in std::this_thread::sleep_for<long, std::ratio<1l, 1000l> > (__rtime=...) at /usr/include/c++/11/bits/this_thread_sleep.h:82
#18 0x0000555555557e99 in main () at /home/rkenyon/repos/ISO11783-CAN-Stack/examples/nmea2000/main.cpp:120
```

ad3154, 2023-01-02:
Hey Riley! Interesting... thank you for reporting this issue. Since the main loop is just sleeping most of the time, it should be pretty straight forward to convert to use this other pattern, where we break out of the loop on a SIGINT rather than aborting within the signal handler itself. According to the references you provided, we're probably hitting UB, which is not ideal

ad3154, 2023-01-02:
Changing the pattern to the one in your reference still results in a double free. This appears to be because `InternalControlFunction` maintains a list of raw pointers, which can mean it has a reference to a deleted object at cleanup time when the program exits. I think this could be fixed by using a smart pointer in the static list, potentially 
[image]

ad3154, 2023-01-02:
Or using a raw pointer I suppose... since I do not know how consumers will use these objects, and because some objects like the virtual terminal client need to hold on to control functions for their entire lifetime, it might make sense to just refactor the interface to use all smart pointers where raw pointers are used currently. This maybe decreases portability somewhat to very resource constrained platforms but is much safer. I am open to opinions on it

ad3154, 2023-01-02:
Still looking at this... if I change the internal static list of ICFs to use shared ptrs, I can't add those pointers to the list in the constructor of the ICF (because the object I'm adding doesn't doesn't have a shared ptr yet technically), which means in order to fully solve this I think either the CAN stack has to take complete ownership of those objects and wrap creation and deletion of them in function calls, (something that might look like `std::shared_ptr<isobus::InternalControlFunction> icf = isobus::CANNetworkManager::CANNetwork.create_internal_control_function(...)` or perhaps we require people to call some function to de-init ICFs before they get deleted so we can clean up first? Still just thinking out loud here

RileyKenyon, 2023-01-02:
Yeah you bring up some good ideas - the easiest would be to have a raw pointer in `main` and could be good for the immediate term. Long term I do like the idea of object ownership in the CAN stack, albeit I may not be familiar with the pitfalls of that approach.

I may not be completely understanding what you are saying about being unable to add ICFs as shared ptrs to the list, but based on your description, would something like `enable_shared_from_this`  be helpful? 
https://en.cppreference.com/w/cpp/memory/enable_shared_from_this

ad3154, 2023-01-02:
I did try inheriting from `enable_shared_from_this`, but unfortunately it has its own issue: 
[image]

ad3154, 2023-01-02:
I actually think I might be able to solve this by replacing the ICF list and the Partner list with arrays, which feels silly but the real issue is that `find` and `erase` with a deleted object I think. If we were just nulling out a pointer there would be no exception... And, we know that there can be no more than 253 of each object, so it's not like, super massive arrays. It's more of a performance hit, but should only happen at startup and shutdown times for most applications... I might experiment with that.

RileyKenyon, 2023-01-02:
Ah bummer on the `enable_shared_from_this`. That is an interesting idea to change it to arrays - also just tried changing to vector and that seems to fix it at least for the NMEA example.

PR #103 (2023-01-03): [#102]: Change how partner and ICF list maintenance is done when members are deleted
When a partner or ICF is deleted, now we will null out its pointer in the list rather than erasing it, which could have caused an exception if the deleted object was a created via smart pointer. Likewise, when a new object is created, we will search first for empty spots in the list before allocating new ones to keep the list as small as possible. Added a trivial unit test to detect exceptions on creation and deletion of ICFs and Partners.

Closes #102
closed 2023-01-03 by ad3154