Open-Agriculture/AgIsoStack-plus-plus #112: Issue while following the Virtual Terminal client tutorial

CRalliNA, 2023-01-06:
Hi,
I have come across another issue whilst following the tutorials in the online documentation.
I am trying to run the code copied directly from the Final Result section of the Virtual Terminal page of the tutorial, but am having another issue with linking the header files. When I run <cmake --build build> I get the error below. It would be great if you could let me know how to resolve it. It seems to be unable to find the header file for a script within "utility" from one of the scripts within "isobus", so thought it might be a wider issue than just the tutorial.

Charlie

<[ 96%] Building CXX object CMakeFiles/isobus_vt_test.dir/main.cpp.o
In file included from /home/charlie/Documents/ISObus/test_VT_client/main.cpp:6:
/home/charlie/Documents/ISObus/test_VT_client/ISO11783-CAN-Stack/isobus/include/isobus/isobus/isobus_virtual_terminal_client.hpp:14:10: fatal error: isobus/utility/processing_flags.hpp: No such file or directory
   14 | #include "isobus/utility/processing_flags.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/isobus_vt_test.dir/build.make:63: CMakeFiles/isobus_vt_test.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:186: CMakeFiles/isobus_vt_test.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
>

GwnDaan, 2023-01-06:
Hey! Thanks for reporting this issue. I suspect you are missing `isobus::SystemTiming` in your CMakeLists.txt which I believe causes this build error.

Make sure it looks something like this: (where VT3ExampleTarget is your executable name)
```cmake
target_link_libraries(
  VT3ExampleTarget PRIVATE isobus::Isobus isobus::HardwareIntegration
                           Threads::Threads isobus::SystemTiming)
```

I don't see anything written about the `CMakeLists.txt` requirements on the online tutorial page which might be a valid TODO. For your reference, a working example you could look at is the [example folder](https://github.com/ad3154/ISO11783-CAN-Stack/tree/main/examples/vt_version_3_object_pool), which had small changes since the tutorial page was published I think.

Let me know if this fixes the issue 😃

CRalliNA, 2023-01-06:
Hi, Thanks for your help, yeah some info on the CMakeLists.txt for the VT tutorial would be great!
I have made that change and I am now getting another error, but saying it can't find the objectPoolObjects.h files, which obviously needs linking from the examples folder somehow, but it is not clear how to do this from the tutorial.

ad3154, 2023-01-06:
Yeah, I think we could probably improve the VT tutorial a bit more. That [objectPoolObjects.h](https://github.com/ad3154/ISO11783-CAN-Stack/blob/main/examples/vt_version_3_object_pool/objectPoolObjects.h) file and its friend [VT3TestPool.iop](https://github.com/ad3154/ISO11783-CAN-Stack/blob/main/examples/vt_version_3_object_pool/VT3TestPool.iop) are tricky to introduce because they are not created during the tutorial but are required, and in a real application would have to be completely created from scratch for your specific application. For the moment I agree with @GwnDaan your best bet is to copy the files from the [examples folder](https://github.com/ad3154/ISO11783-CAN-Stack/tree/main/examples/vt_version_3_object_pool) but I think we should add to that tutorial:

* CMake
* The content of objectPoolObjects.h
* Download link to the test object pool with instructions on how to use it

CRalliNA, 2023-01-06:
Hi Adrian,
I've got it going now by copying the two files from the examples folder, but had to change two lines of the main.cpp file from the tutorial to make it work, it looks like the naming convention had been changed.

I changed line 134 and 135  of the tutorial from :
<        TestVirtualTerminalClient->RegisterVTButtonEventCallback(handleVTButton);
        TestVirtualTerminalClient->RegisterVTSoftKeyEventCallback(handleVTButton);>
to 
<        TestVirtualTerminalClient->register_vt_button_event_callback(handleVTButton);
        TestVirtualTerminalClient->register_vt_soft_key_event_callback(handleVTButton);>

and it now appears to be working.
Thank you

ad3154, 2023-01-06:
Excellent! We'll make sure to go in and fix that issue as well. The tutorials are fairly new and we haven't had a ton of feedback yet, so your feedback is very much appreciated.

ad3154, 2023-01-07:
I've made some changes to the VT tutorial in #113 hopefully they help clarify things and resolve the issues you reported. Feel free to review them if you'd like and make further suggestions, otherwise I'll let the PR stay open for a day or so to see if it gathers any comments then get it merged in. The changes are live already though on the website.

PR #113 (2023-01-08): [#112]: Update VT tutorial with CMake, Fix omissions and a couple outdated function names
* Replaced ${CMAKE_THREAD_LIBS_INIT} with Threads::Threads (See #54)
* Added CMake instructions to VT tutorial
* Fixed `RegisterVTButtonEventCallback` and `RegisterVTSoftKeyEventCallback` should be snake_case
* Clarified use and instructions around `objectPoolObjects.h` and `VT3TestPool.iop`
* Added a section about the other callbacks, like string value events
* Added some more references to the API docs

Closes #112 
Closes #54
closed 2023-01-08 by ad3154