Open-Agriculture/AgIsoStack-plus-plus #121: Bug in receiving messages via extended transport protocol

CRalliNA, 2023-01-10:
**Describe the bug**
Hi, I've been doing some fiddling to get my head around the software and had the CanStack hooked up to a Virtual Terminal Client and was trying to receive the VT object pool from it. In the process of trying to receive these messages over ETP, I came across what I think is an issue, please correct me if it is actually desired behaviour though. The first thing I noticed was that the payload of the message did not match what I was expecting. I verified this with candump and the start of the payload is definitely different - this lead me to thinking that the payload had been cut short with the bytes at the start having been removed.  I then noticed that the length of the message provided by `CANMessage->get_data_length();` is the hexidecimal value of the expected decimal number. This can be seen in the Extended Connection Mode RTS message in the candump as the number returned is exactly the same as the bytes in the message not the decimal version of it that I would have expected. (If it is meant to be in hex please let me know, I could be barking up the wrong tree). I had a look in the can_extended_transport_protocol.cpp file and it looks like the issue must be at line 94 where the value for the size of the new session is calculated from the body of the Extended Connection Mode RTS message. However, I can't work out exactly what is causing the issue. 

**Supporting Documentation**
If your issue is CAN behavior related, it may be helpful to have a CAN trace of your issue that clearly shows the undesirable behavior. Please attach one if you think it might help, or if requested. Try to capture address claiming in addition to the bad behavior if possible. We prefer files captured with `candump -l` or `.asc` ascii format log files.  Binary Vector files `.blf` will not be accepted. 

This is the candump of the process of the object pool transfer
[candump-2023-01-10_183013.log](https://github.com/ad3154/ISO11783-CAN-Stack/files/10385769/candump-2023-01-10_183013.log)

However, this is the first 100 bytes output from the Canstack software:
1 1 1 1 1 1 1 1 2 1 1 1 1 1 1 1 3 1 1 1 1 1 1 1 4 1 1 1 1 1 1 1 5 1 1 1 1 1 1 1 6 1 1 1 1 1 1 1 7 1 1 1 1 1 1 1 8 1 1 1 1 1 1 1 9 1 1 1 1 1 1 1 a 1 1 1 1 1 1 1 b 1 1 1 1 1 1 1 c 1 1 1 1 1 1 1 d 1 1 1
Printed with:
`std::vector<std::uint8_t> messageData = CANMessage->get_data();
for ( int i = 0; i < 100; i ++){
      std::cout << std::hex << uint32_t(messageData.at(i)) << " ";
}
std::cout << std::endl;
`

As you can see in the candump file the Extended Connection Mode RTS message sent is : 18C81C80#145648020000E700 - bytes 2-5 give the total number of bytes to be sent in little endian format  - 0x024856 which would give 149,590 in decimal. However, the output for the size of the message received from the CanStack  the code below) was 24856 - the same as the hex value in the message.
`std::cout << "Transfer message Data length = " << CANMessage->get_data_length()  <<std::endl;`

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Compiler GCC 9.4.0
 - CAN Driver Socket CAN

**Additional context**
Add any other context about the problem here that you think might help us recreate or otherwise address your issue if applicable.

ad3154, 2023-01-11:
Hello again!

ETP Rx is probably the least tested part of the transport layer so I appreciate the feedback.
I've been able to recreate/verify that there is an issue with the message contents, so I'll dig into that. Something's definitely strange there...

I haven't been able to recreate the size issue though. I sent the stack a a 6000 byte message from one PC, and received exactly 6000 at the other, for example, - so at this point I suspect it might just be lingering settings on your `stdout` converting it to hex when its displayed.

I've also noticed that unaligned (so, most) EDPO messages can cause the stack to erroneously claim that the DPO packet max didn't match the CTS message we sent, which is not true, so I have to fix that as well!

ad3154, 2023-01-11:
I think I have these two issues fixed in this branch: https://github.com/ad3154/ISO11783-CAN-Stack/tree/bugfix/etp-rx-issues

If you have time, I'd appreciate if you could re-try your ETP reception on this branch and let me know the results.

```
git fetch --prune
git checkout --track origin/bugfix/etp-rx-issues
```

CRalliNA, 2023-01-11:
That's brilliant, thanks. Will have a look now.

CRalliNA, 2023-01-11:
I have tried it again on the new branch, and now get the output I expected from the ETP message. I still get the message length in hex, but I think I was barking up the wrong tree.

PR #122 (2023-01-12): [ETP]: Fix ETP Rx message contents not matching the expected content
* Fixed an off by 1 error in parsing the protocol bytes out of each ETP frame
* Fixed the data offset into the resulting message buffer would roll over due to it using the wrong counter
* Fixed an erroneous log message by setting the expected ETP packet count properly when we CTS less than 255 frames, usually at the end of a session.

Closes #121
closed 2023-01-12 by ad3154