Open-Agriculture/AgIsoStack-plus-plus #123: Auxiliary input (type 2) support for VT application

GwnDaan, 2023-01-12:
Hey! Would support for auxiliary/physical keys of the stack be something that is planned to be implemented? I don't really have prior experience implementing this. However, from Part 6 (and Figure J.8) of the standard, I gather that the following should be implemented:

For communication to/from VT:
- [ ] Auxiliary assignment type 2 command
- [ ] Auxiliary assignment type 2 response
- [ ] Preffered assignment command
- [ ] Preffered assignment response
- [ ] VT Status (Learn Mode) <- not sure if necessary

For data from auxiliary input device:
- [ ] Auxiliary assignment type 2 status message
- [ ] Auxiliary assignment type 2 maintenance message

I guess this could be another callback that the application implementing the stack could register?

ad3154, 2023-01-12:
I have also not implemented this functionality before, and AUX-N (Auxiliary assignment type 2) is a bit tricky for us because in a multi-VT environment, you may have your working set loaded to a VT with function instance 1, but have aux assignments with the VT with function instance 0. In those cases we'd need to send the working set maintenance message to both VTs and handle messages from them both at the same time. I think we should add these functions, but it may require us to think through some interesting challenges like that, as well as how we want to structure it all.

GwnDaan, 2023-01-12:
Ah, I see. I agree with thinking through how we want to structure this all. Especially since there are still quite a few features left for the VT to implement now (and possibly more in the future when even newer VT versions are published).

For keeping it all easier to maintain, we might want to split some features of the VT into separate files? Doing large refactorizations is probably better to do as soon as possible, as there are relatively few applications that have implemented the stack yet. I feel like splitting the files helps for optimizing the stack easier later on, as well as not losing track of where specific code is located - the last one particularly useful for people (like me) who are not experts at using shortcuts/keybinds in editors haha).

This might even be worth to create a separate issue for, for better overview for what's todo? If we plan to restructure, I think this feature should be postponed until that is finished.

ad3154, 2023-01-12:
Perhaps... I do worry about splitting it up too much, as some of the state information and VT info would either have to be duplicated, or we'd have a circular dependency, or they'd have to be `friend` classes or something, none of which I super love. Maybe this auxiliary stuff wouldn't be too tough though, as a first attempt at it, I think we could have an optional parameter or setting for the AUX-N display control function (if the primary one is not function instance 0), and then just provide callbacks to the user. A more "managed" solution though might be to keep track of the assignments, NAMEs, and all the other associated metadata for the user, but then I think we'd have to require the use of files or something to properly manage all that and save it, which I also don't love as not all devices have filesystems. All our filesystem stuff is optional today.... Just thinking out loud. I think a good first attempt at it would be as you say, and to provide callbacks.

ad3154, 2023-01-12:
Oh and for the other VT features we're missing, maybe we should create issues for those so we can group them into some kind of milestone, so we can track its completeness?

GwnDaan, 2023-01-12:
Yeah... that's true. I definitely stand for keeping the filesystem optional, so instead we could forward this implementation to the application itself? Like making it callback based or using an abstract base class. The storing and loading of the "preferred" assignments could be optional I think. Same goes for factory default assignments of auxiliary functions, which can also be empty. I (also) think that we should get the fundamental parts implemented for now - so without these two optional features, but keeping them in mind. And then see how it goes from there.

> Oh and for the other VT features we're missing, maybe we should create issues for those so we can group them into some kind of milestone, so we can track its completeness?
Sure thing! I'll create them for the ones I come across, or perhaps I'll read over the standard someday if I have extra spare time. I think there will be quite a few issues open if we raise them for everything that's left. But then we don't have to get bored 😛

ad3154, 2023-01-13:
Yep that approach sounds reasonable to me I think!

GwnDaan, 2023-01-13:
Alrighty, I can give this a shot if that's okay with you?

ad3154, 2023-01-13:
Absolutely, if you're interested in taking a crack at it, that would be super!

GwnDaan, 2023-01-14:
I started implementing in https://github.com/GwnDaan/ISO11783-CAN-Stack/commit/5eeba6401c732750a28138f3324550a79310c128, but I got kind of stuck [here](https://github.com/GwnDaan/ISO11783-CAN-Stack/commit/5eeba6401c732750a28138f3324550a79310c128#diff-e5352da9d8cf38470aa4b854fccdb6c83520f146f6f52b74898d038f3a698312R2988-R2994) where we have to call the callback itself. What would be the best arguments to pass on to the callback function? Or would an abstract class suit this better? 

The problem is that there are 15 different function/data types for auxiliary input (Table J.5 of the standard). Each with different ways of interpreting `value1` and `value2`. We could just pass the "raw" values in the callback, but I don't really like that the application itself has to implement the definition:
```c++
typedef void (*AuxiliaryInputCallback)(std::uint16_t inputObjectID, std::uint8_t functionType, std::uint16_t value1, std::uint16_t value2, VirtualTerminalClient *parentPointer);
```
Though I also don't want to register 15 different callbacks for all of the possible function types. Any ideas?

ad3154, 2023-01-14:
Ah yeah I see, technically the field allows up to 32 function types (many are reserved).

That is tricky, I agree having that many callbacks seems bad. We may just want to decode that function to be some enumerated value, which at least does a little of the parsing for users. 

Like
```
enum class AuxiliaryFunctionTypeTwoType
{
  BooleanLatching = 0x00,
  ...
};
```

Beyond that though I am not sure....

closed 2023-01-17 by ad3154