Open-Agriculture/AgIsoStack-plus-plus #134: Add support for Inno-Maker USB2CAN

ad3154, 2023-01-16:
I have an inno-maker USB2CAN that has an open source driver here https://github.com/INNO-MAKER/usb2can

This adapter has a native Linux driver (Socket CAN) but it would be fairly trivial to add support for Windows, which would be convenient given the adapter's low cost. Can maybe also add MacOS but I do not have a Mac to test on, so we would need someone to help test that if we add it.

GwnDaan, 2023-01-26:
Oeh nice, I see you already started on this. I just got myself this little guy.

[image]

GwnDaan, 2023-01-27:
Hmm, I can take over the development of this if you like? I have some extra free time this weekend and I'm curious to see this thing in action.

ad3154, 2023-01-27:
Sure, I'd appreciate it

PR #160 (2023-01-29): [Core]: Add InnoMaker USB2CAN plugin
After struggling for a while, here is the USB2CAN plugin. The documentation for it is minimal lol. I was able to run the VT example successfully while using this plugin in the end though.

Note: Make sure to add the `InnoMakerUsb2CanLib.dll` file to the directory you are running the executable from. E.g. when building the `vt_version_3_object_pool` example from top level in debug mode. Add the dll file to the `/build/examples/vt_version_3_object_pool/Debug/` folder.

Closes #134
closed 2023-01-29 by ad3154