Open-Agriculture/AgIsoStack-plus-plus #136: Aux-N persistent assignments support

GwnDaan, 2023-01-18:
The stack supports now basic Aux-N assignments for the VT. I think we want to have the stack save these assignments persistently and load them in on next startup. This is also what the stack requires/wants from the working set. 

To make this implementation not depend on any particular storage type, my idea was to make the storage part abstract and let the application decide which one they want. We can possibly also provide some of these implementations of the abstract storage part - For example the Linux and/or Windows filesystem, ESP flash storage, etc...

**Related TODOs**:
Auxiliary assignments should be loaded (e.g. from file/flash/cloud etc...) and then published to the VT after it loaded the object pool:
https://github.com/ad3154/ISO11783-CAN-Stack/blob/22094dd22a460e46d8778d9a40c26ac5d529a3d0/isobus/src/isobus_virtual_terminal_client.cpp#L2318
After these auxiliary assignments have been published to the VT, we expect to get an approval/denial. If the preferred assignments command is approved (and therefore the assignments are implemented) by the VT, we can have to load them in on our end as well:
https://github.com/ad3154/ISO11783-CAN-Stack/blob/22094dd22a460e46d8778d9a40c26ac5d529a3d0/isobus/src/isobus_virtual_terminal_client.cpp#L2863
If the VT operator assigns an auxiliary input to our function, and the VT wants us to save this assignment persistently, we have to store it as well. (e.g. to file/flash/cloud etc...)
https://github.com/ad3154/ISO11783-CAN-Stack/blob/22094dd22a460e46d8778d9a40c26ac5d529a3d0/isobus/src/isobus_virtual_terminal_client.cpp#L2896

Extension of #123

GwnDaan, 2023-01-21:
Working on this in [feature/persistent-aux-n](https://github.com/GwnDaan/ISO11783-CAN-Stack/tree/feature/persistent-aux-n). I implemented new layers for the storage part, but I'm not sure if I like it this way. Any feedback is welcome

gunicsba, 2026-11-05:
@GwnDaan with #655 merged, the VT client loads and stores preferred assignments through `set_auxiliary_preference_callbacks()` and resets them when the model identification code changes. The VT server side is in #836 and #846. Is anything left here, or can this be closed? I'll leave closing it to you.