Open-Agriculture/AgIsoStack-plus-plus #165: Strange VT related crash on embedded ESP32-S3 application

GwnDaan, 2023-02-01:
**Describe the bug**
I used the current main branch with the 'vt-version3-object-pool' and removed the `[] { return true; }` part which was introduced in this commit: https://github.com/ad3154/ISO11783-CAN-Stack/commit/046461bfb2276d9fb3a107de2d3076d862666301 but apparently blocks the application from reaching the watchdog timer and results in the watchdog timer raising exceptions.

When running the slightly adapted 'vt-version3-object-pool' example:
- It runs fine when you don't initialize the VT, so the vt doesn't update but the other stuff such as address claiming still works
- With the VT initialized it runs until state=13 (WaitForGetVersionsResponse) in the `update()` function and somewhere immediately after that the application crashes.

**Supporting Documentation**
The latest branch that I was able to get it all working fine was at 5b159f69596254c1a879b79388474ff45f204739. Somewhere since then we must have introduced something which badly corrupts memory I think?

**Environment (please complete the following information):**
 - OS: ESP-IDF
 - Compiler: PlatformIO
 - CAN Driver: TWAIPlugin

ad3154, 2023-02-01:
Out of curiosity, does it work if you bypass the versioning by removing the version from the `set_object_pool` ?

GwnDaan, 2023-02-01:
Uhmmmm... very vague stuff. It fails now anywhere from 5 seconds to 30 seconds from start of uploading the pool and it's taking ages to upload a pool compared to that "stable" commit https://github.com/ad3154/ISO11783-CAN-Stack/commit/5b159f69596254c1a879b79388474ff45f204739. Like not even 10% uploading in 30 seconds compared to the whole pool uploading within 20 seconds before. Any ideas?

ad3154, 2023-02-01:
Hmmmmmm.... Have you tried any commit in between? I would maybe suggest 0eeba9fa18454f43981e82020739395bc0c5dfa4 as I have seen some _interesting_ things happening since #163 with some of the mutexes... maybe you are getting deadlocked.

ad3154, 2023-02-01:
For example, on Windows with PEAK, if the bus is down when I call `start` it murders the mutex while locked which causes a nasty exception deep in the Windows files....

GwnDaan, 2023-02-01:
Another quick update; when I add 1 extra logging statement in `VirtualTerminalClient::process_rx_message` it crashes already after reaching state=2 (SendWorkingSetMasterMessage), these are the changes I made btw: https://github.com/ad3154/ISO11783-CAN-Stack/commit/c20da6fdc0ed1f1d11c7cf05a3dbbf42dd7c3cf8

GwnDaan, 2023-02-01:
Good tip! I'll try some branches in between while I am at it

GwnDaan, 2023-02-01:
- https://github.com/ad3154/ISO11783-CAN-Stack/commit/0eeba9fa18454f43981e82020739395bc0c5dfa4 still same issue, but instead of crashing it freezes now
- https://github.com/ad3154/ISO11783-CAN-Stack/commit/46324fcdc93964c916c260d3bac6b6f3bca8f682 no crashes but still extremely slow upload speed, so slow that we even receive aborts from the VT
- https://github.com/ad3154/ISO11783-CAN-Stack/commit/c9a4ef439de7c64a69a7ab9cf4b2f53e88b8b97f still slow but no aborts and it was able to upload to 100% after ~8 mins but with an error in the object pool?
- https://github.com/ad3154/ISO11783-CAN-Stack/commit/5b159f69596254c1a879b79388474ff45f204739 which was the stable branch is still super slow lol?

I then started testing on my personal project, where it was able to upload the same pool in approx 1 min, which is still pretty slow and it had to retry a few times because it received aborts. I guess our TWAIPlugin has never been really reliable nor fast.

So I guess I will do some more testing/bugfixing tomorrow:
- [ ] Compare the current vt_version3_object_pool example with the previous version/my project and try to find the bottle neck because it is now about 8x slower.
- [ ] Try to fix the unreliable upload with the TWAIPlugin.

Still the origin of the crash is somewhere introduced between https://github.com/ad3154/ISO11783-CAN-Stack/commit/0eeba9fa18454f43981e82020739395bc0c5dfa4 and https://github.com/ad3154/ISO11783-CAN-Stack/commit/46324fcdc93964c916c260d3bac6b6f3bca8f682. To compare: https://github.com/ad3154/ISO11783-CAN-Stack/compare/46324fcdc93964c916c260d3bac6b6f3bca8f682...0eeba9fa18454f43981e82020739395bc0c5dfa4

GwnDaan, 2023-02-01:
Bruh... found why it's so slow 😂
```c++
CANHardwareInterface::set_can_driver_update_period(100);
```
I forgot that I had set it so slow haha, I changed it when it was crashing to check if that caused the crashes, but forgot to change it back.

ad3154, 2023-02-03:
Hahaha, yeah that would do it! 😂

GwnDaan, 2023-02-08:
### Fixed:
The issue was, as far as I understand, that the main update thread of the HardwareIntegration part of this stack did not have enough memory available to do all it's tasks and so it crashed, therefore it was needed to increase the [CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-pthread-task-stack-size-default) value of the ESP32 in the menuconfig. (I increased it from 3072 to 6144).

We could add this to the FAQ one day if there are more people having this issue... Cus I found it damn hard to find the cause of the crash

closed 2023-02-08 by GwnDaan