Open-Agriculture/AgIsoStack-plus-plus #192: Interfering internal CF addresses

GwnDaan, 2023-02-26:
**Describe the bug**
When creating two internal CF addresses with the same preferred address, they will both continue using the same address. This causes information for one devices to be handled by both and leading to undefined behavior. This bug occurred by running two executables, though it might also happen within the same executable?

**How to replicate**
Run the VT "auxiliary function" and "version3 object pool" examples simultaneously, with both of them preferring the same internal address.

ad3154, 2023-02-26:
I think the real issue here might be that your 2 executables have the same ISO NAMEs, which is not allowed, and we have no way to prevent across 2 executables. In theory, the stack should resolve them to different addresses even if they share the same preferred address if their ISO NAME is different. If you change one of the NAMEs and it still does this, then we'll have to take a closer look...

GwnDaan, 2023-02-26:
Hmm I just tried to define unique NAMEs for them, it seems to solve half of the issue. With some specific combination of NAMEs they still acquire the same address. To reproduce:

First running the "aux input" and "aux functions" example together on [this branch](https://github.com/GwnDaan/ISO11783-CAN-Stack/tree/bug/claiming-same-address). It works fine and they both claim separate addresses.
[image]
The difference between the NAMEs for them is:
```cpp
/// aux_inputs NAME: 
TestDeviceNAME.set_ecu_instance(1);
TestDeviceNAME.set_function_instance(0);

/// aux_functions NAME:
TestDeviceNAME.set_ecu_instance(0);
TestDeviceNAME.set_function_instance(1);
```

But when running "aux input"/"aux functions" together with "version3_object_pool", in both cases I get the same bug:
[image]
The difference of "version3_object_pool"  over the others is:
```cpp
TestDeviceNAME.set_ecu_instance(0);
TestDeviceNAME.set_function_instance(0);
```

Any idea what this cause of this might be? Or is this intentional for specific NAME combinations?

ad3154, 2023-02-26:
Well, two control functions with different NAMEs should never collide, so there might be some issue there. I'll probably need to try and reproduce it to see what's going on. If possible, can you provide a `candump -l` trace of this?

GwnDaan, 2023-03-10:
Sorry, it took a while before I had the VT back on my desk to test with again. Busy times on the farm already 💩. Let me know if you can find anything, I can't really make anything of it. 

[candump-2023-03-10_224131.log](https://github.com/ad3154/Isobus-plus-plus/files/10946355/candump-2023-03-10_224131.log)

The corresponding consoles (Note that I ran the right one (aux functions) before the left one (virtual terminal), the other way around they claim just fine):
[image]

GwnDaan, 2023-03-10:
Oh, here is the log of the other way around:
[candump-2023-03-10_224756.log](https://github.com/ad3154/Isobus-plus-plus/files/10946404/candump-2023-03-10_224756.log)

With the consoles:
[image]

ad3154, 2023-03-10:
Excellent, I will try and take a look at this over the weekend

ad3154, 2023-03-11:
Ahhhhh I see. What you are seeing is this:
[image]

Basically, since they both have 0x1C as their preferred address, the one with the lower ISO NAME wins according to the standard. The aux input example moves to address 0x80 since it failed arbitrating for 0x1C. This is normal, as expected behavior as far as I know.

ad3154, 2023-03-12:
I think we may want to add logging statements to the address claiming state machine to make this more clear, because in the logs it is tough to tell what is happening without the candump

PR #249 (2023-06-17): [Core]: Improved lost CF pruning
* Added active control function pruning so that if a control function fails to claim after a request for the address claim PGN, we move that control function to the inactive list and invalidate its address. 

This was always meant to be a part of how we manage address claiming, but for whatever reason got left behind. This greatly increases the reliability of checking a control function's address being valid.

* Fixed a random crash in unit tests.
* Added additional logging to the address claim state machine to more clearly describe what is happening.

Closes # [... cut]
closed 2023-06-17 by ad3154