Open-Agriculture/AgIsoStack-plus-plus #193: Rust Support

JannesBrands, 2023-02-26:
Hey there!
First of all, a big thanks for this project. ISOBUS for everyone was and is unfortunately still a problem. I have worked with the software library of OSB-Ag and CCI myself and I am glad to see such a project, which offers a reasonable option to write an ISOBUS application.

I am currently working with Rust and would like to implement the majority of my ideas and projects with Rust in the future. The advantages are well known and I think it would be cool if this library could find a way to be one step ahead of other proprietary ISOBUS libraries. 

I know that a complete rewrite of this project in Rust would be a lot of work, but maybe we could keep this as an idea/thought for now, since this project is not too far advanced yet and it would be more conceivable at this point than if this project fulfills all the functionalities of the 11783 standard.

Translated with www.DeepL.com/Translator (free version)

ad3154, 2023-02-27:
Hey there!

I have been wanting to learn Rust, and you're not the only person to ask, so I'm at least a little interested in developing a Rust version in another repository. Most of the applications I need this library for are written in C++ today though, so I'd like to get this C++ version at least to some kind of 1.0 stable state before really considering changing gears or splitting my attention with a Rust version. For what it's worth, I don't think that's particularly far away, as my plan has always been to finish the task controller client and the file server client, do some tidying up, then call that 1.0. Then we can switch from major features more towards full ISO/AEF conformance and bugfixes in subsequent updates, which will allow me some more freedom in how I spend my evenings, haha.

So long story short, yes, I'd like to make a Rust version too when I can if life allows. I think more people might contribute as well if it were Rust based, which would be nice. But my priority at least is the C++ version until it's stable and has those last 2 major features working well.

GwnDaan, 2023-02-27:
Hi!

I thought I'd also join in here. I've been seeing more and more Rust coming by lately, and as far as I've seen it looks great. Especially the on compile-time memory safety attracts me. Like @ad3154  says, we'd probably be best of releasing a stable version first before switching gears. But I'm definitely down for working on a Rust version as well. I saw that [@Thom-De-Jong](https://github.com/Thom-de-Jong) had started working on a Rust version a while ago, but it seems he already pulled it off GitHub. 

I'm pleasantly surprised how many people are interested in ISOBUS, and it seems to me that perhaps the most powerful thing to do is to bring it all under a common GitHub organization. I know I already touched on this in #175 but it seemed useful to bring it up here as well.  Maybe then we'd have to find another name instead of isobus-plus-plus as well, I don't know.

ad3154, 2023-02-27:
Yeah I agree with everything @GwnDaan says. Maybe we do need a more general organization name, which can house the Isobus++ branded stuff, plus whatever else we come up with, and is more open to just people in general who are interested in ISOBUS. I also don't really want my personal account to be the home for all this stuff haha. I am open to ideas on naming for sure, I am not very good with names.

ad3154, 2023-03-15:
Converting this issue to a discussion. We still should come up with an organization name so we can place something like a Rust implementation under that grouping.

closed 2024-03-09 by JannesBrands