Open-Agriculture/AgIsoStack-plus-plus #200: TC Client Example

bjoernpoettker, 2023-03-05:
Hello,

I have tried to run the TC Client example - for the first run it worked and everything was loaded on my Fendt. When I started the example another time, I can't get it run. 
[image]
All other examples work - the VT is loaded, AUX-N works.

Does somebody know what I made wrong?

I build the stack in VS2022 on Win10, use the USB2CAN-X2, the Tractor has the 10.4" NT01 ter,minal and the newest Software (V 790/843)

ad3154, 2023-03-05:
Hmmm... TC client is still _very_ experimental, so keep that in mind when trying to use it in the current state (like, I haven't added the theshold and measurement commands yet, which is an important component. 

But, in this case, I did find a bug very quickly thanks to your log output that you included. I have pushed a fix here: https://github.com/ad3154/Isobus-plus-plus/pull/195/commits/3c0f1c37552325a523cd9988ec3f0680573f762a

The reason it worked OK the first time was because we didn't request the localization label because the structure label didn't match the one in the example. On the second time, you hit the bug because in the message processing I was checking the wrong state when handling the localization label. 

So yeah, if you pull the latest TC branch hopefully it will work better, but I wouldn't expect fully working section control until I finish the last missing parts.

ad3154, 2023-03-05:
Let me know if this works OK and I will close the issue out.

bjoernpoettker, 2023-03-05:
Hi,
I have tried your changes - now it works when I restart!
I know that this is very experimental but I have tried to write my own TC client years ago but I wasn't able to get it work.

If it helps you I can help to test the code on some other devices - I have besides the Fendt terminal some CCI (200, 800, 1200) and a JD 4640 display to test!

closed 2023-03-05 by bjoernpoettker