Open-Agriculture/AgIsoStack-plus-plus #206: Question about internal/partner functions (I know this isn't a support forum).

Y-Less, 2023-03-09:
First - is there some better support forum for basic usage questions?  Forums/discord/e-mail etc?  I know that this isn't what Github issues are meant to be used for sorry.

Secondly.  I'm a bit lost as to the purpose of internal functions when using partnered functions.  According to the documentation you pass an internal function as the source when sending messages to partnered functions.  But then responses from the partner get sent to the partnered function directly after using `add_parameter_group_number_callback`.  So it seems like the internal function is the original source, and the partnered function is both the destination and the response receiver.  Why is the internal function (which was the source of the original message) not the recipient of the reply; why is it not even referenced in the response callback; and would sending a new message from the response callback need to acquire that internal function again from somewhere?

In short - why is it used for outbound messages, but ignored for their responses?

ad3154, 2023-03-09:
You could [start a GitHub discussion](https://github.com/ad3154/Isobus-plus-plus/discussions)? The community is pretty small right now, so we're mostly just here on GitHub right now. I think I might be able to convert issues to discussions as well if they are more like clarification stuff.

So, you are right in thinking that the internal control function is source to send messages from, and the partner is a destination. This concept is reversed though when receiving destination specific messages from a partner.
You can also of course pass in `nullptr` as the partner to send a broadcast (and indeed this is required for some PGNs).

The "response receiver" as you call it is whatever callback you add via the partner class (or using `add_global_parameter_group_number_callback`), and generally the context is completely up to you, it doesn't have to be tied to the partner in any way really except in the implicit sense that you are telling the network manager that you care about messages from that partner. The CAN message class you get from the network manager actually contains pointers to both as well! The destination control function (`get_destination_control_function`) contains a pointer to your internal control function which is the destination of that message. Though, if you want to use it as in internal control function again in there, you'd want to check the type and cast it appropriately.

However, I think what you really might be asking is "Why can't you pass in an internal control function into `add_parameter_group_number_callback` (or something like that) to only get callbacks for one specific pair of control functions?" That is an excellent question that I did not think of until now to be honest. Currently the behavior in the network manager is that it will call your callback when _any_ internal control function is the destination (or when the destination is a broadcast if you used `add_global_parameter_group_number_callback`) and then it's kind of up to the consumer of the callback to determine what to do based on the internal control function in the callback. There is an argument to be made I think for adding a specific internal control function to the registration, that seems like a good idea to me. If this addresses your question, I think we can re-title this issue and use it for that purpose if that's OK with you.

On a side note, I will also say that it's not really a good idea to do any work inside the callback besides parsing and saving data. Blocking the network manager means blocking _everything_ including ongoing protocol sessions, some of which require timely interaction. The best way to handle this is to save what you need, and do the real work in your main application rather than in the network manager's thread as much as possible.

Y-Less, 2023-03-10:
> However, I think what you really might be asking is "Why can't you pass in an internal control function into `add_parameter_group_number_callback` (or something like that) to only get callbacks for one specific pair of control functions?" 

Yes, I think that's an excellent distillation of the core issue, when I didn't really know enough to be able to say it.

> If this addresses your question, I think we can re-title this issue and use it for that purpose if that's OK with you.

Sure.

> The CAN message class you get from the network manager actually contains pointers to both as well!

I hadn't noticed this, so in the interim at least the information is there and usable, which is all I really wanted even if there may be simpler ways to do it in the future.

> On a side note, I will also say that it's not really a good idea to do any work inside the callback besides parsing and saving data. 

Noted.

Thank you very much for the information/clarifications/help.

PR #250 (2023-05-10): [Core]: Add internal CF filtering option for partner CF callbacks
This allows users to register callbacks for messages from a partner that are filtered by destination internal control function.

This essentially helps you only get callbacks for messages destined for one specific ICF if you have multiple, making a multi-ECU application easier/nicer.

Closes #206.
closed 2023-05-10 by ad3154