Open-Agriculture/AgIsoStack-plus-plus #210: Callback for partner function status changes

Y-Less, 2023-03-12:
**Is your feature request related to a problem? Please describe.**
If a partner function takes a long time to come up, or goes down then comes back on a different address, I want to be able to send some messages to it straight away.

**Describe the solution you'd like**
A callback to inform when a partner function changes status, either comes up, goes down, changes address, and I'm sure there's more:

```c
enum partner_control_function_status
{
    up,
    down,
    moved,
    // etc.
}

void status_callback(PartnerControlFunction * partner, partner_control_function_status status, void *parentPointer)
{
}

CANNetworkManager::CANNetwork.add_partner_control_function_status_callback(partner, &status_callback, nullptr);
```

**Describe alternatives you've considered**
I did check the code, but the best I could find was polling `get_address_valid`, and that only gives the initial bring-up, not later changes.

Y-Less, 2023-03-12:
Maybe the `partner` parameter to whatever register function is used isn't even needed if you just want the callback called for them all.  Though that would probably result in some sort of lookup in the function instead a lot of the time.

Y-Less, 2023-03-12:
Even polling can miss changes if the partner goes down and up again fast enough.

ad3154, 2023-03-12:
I think we could add at least up/down notifications. Makes interacting with it a little more event driven. Changing address should be completely obscured from the user I think, as the control function abstraction is supposed to make that unimportant information. That being said, I suppose if you wanted to display an address on the VT, and you wanted an event for knowing when to update that object, maybe then it would be useful? Not sure... I guess, do you have a particular use case for that one in mind?

Side note: The only time a control function would go "down" is if address claiming happened again, and that control function did not claim (implying it is offline). If `get_address_valid` is not providing that feedback, then there may be a bug!

PR #308 (2023-09-02): [Core]: Added a way to get callbacks when any CF goes online or offline
### What's New:

* This addresses issue #210 and provides a way to get callbacks from the network manager when a control function changes state.
* Overall, this reduces the need to poll the state of your partners in a consuming application.
* Also fixed some typos.

Closes #210
closed 2023-09-02 by ad3154