Open-Agriculture/AgIsoStack-plus-plus #221: VT send_change_size_command doesn't consider scaling factor

GwnDaan, 2023-03-22:
By adding the dynamic object pool scaling in #164, we now have functions that don't take into account the scaling factor. Such as `send_change_size_command`:
https://github.com/ad3154/Isobus-plus-plus/blob/44a2941ca23f50f0a7ed07b4f74b460c91db0815/isobus/include/isobus/isobus/isobus_virtual_terminal_client.hpp#L575-L582

I can think of 2 possible solutions:
- Make a public getter for the scaling factor, and let the user implement the scaling themselves
- Automatically scale the dimensions inside the `send_change_size_command`

I'd prefer the latter, but both might be doable as well with a parameter option in the `send_change_size_command` function. E.g. `automaticallyScaleDimensions`.

ad3154, 2023-03-22:
Ah, good thinking! I think I'd like to allow both if possible: setting an absolute size and auto-scaling. So maybe we should add a default parameter for scaling to some of these functions? Plus add a function to get the scale factor?

GwnDaan, 2023-03-22:
> Ah, good thinking! I think I'd like to allow both if possible: setting an absolute size and auto-scaling. So maybe we should add a default parameter for scaling to some of these functions? Plus add a function to get the scale factor? 

Okayy sounds good! I can work on this

GwnDaan, 2023-03-23:
We'd also need to take into account #223, as someone may also want to scale some object in a soft key using a command.

gunicsba, 2026-11-05:
Related: #833 scales objects shown on soft keys with the soft key factor during upload (#223). For this issue, `send_change_size_command` would need the same choice at runtime. `get_soft_key_scaled_object_ids()` from #833 gives the set of objects that use the soft key factor.

closed 2026-11-10 by gunicsba