Open-Agriculture/AgIsoStack-plus-plus #234: Expand VT Client Tutorial

ad3154, 2023-04-02:
**Is your feature request related to a problem? Please describe.**
We sometimes get questions about how some VT client interface functions work, and only really show simple button interactions in the VT client example.

**Describe the solution you'd like**
I'd like to see the VT client expanded with additional kinds of interaction with the VT, such as changing strings, moving things, using object pointers, gauges, and things like that. This may help users of the project more easily create applications with the stack by having a broader set of working examples.

**Describe alternatives you've considered**
There's only so much documentation can do in this regard, but we could also consider expanding the tutorial documentation to more clearly list the requirements and pitfalls of certain interactions with the VT.

bjoernpoettker, 2023-05-17:
Hi guys,

I tried the seeder example but I got an error after uploading to my Fendt.

In the console it logs the following error:

 [←[1;31mError←[0m][VT]: Error in end of object pool message.Faulty Object 6000 Faulty Object Parent 65535 Pool error bitmask value 9

ad3154, 2023-05-21:
I've pushed an update to change the object pool to version 3. This example is still a work in progress, as I am adding an ISOBUS odometry interface to read speed information before fully completing it.

closed 2024-02-12 by ad3154