Open-Agriculture/AgIsoStack-plus-plus #238: VT client unable to reset connection with VT server

GwnDaan, 2023-04-15:
**Describe the bug**
The VT seems to be having a hard time resetting the connection with my Mueller TOUCH800:
[image]
Has anyone else experienced this issue with other terminals as well? I'll try to make a candump sometime soon

**Environment (please complete the following information):**
 - OS: ESP32-S3 (ESP-IDF)
 - CAN Driver: TWAI
 - VT Server: Mueller TOUCH800

ad3154, 2023-04-15:
The get memory message is the first message in the state machine that we send to the VT where we expect a response, so it's also the most common place to get stuck if there's some underlying communication issue on the bus or with the hardware plugin. A candump would definitely be helpful when you have time.

GwnDaan, 2023-04-25:
I managed to make a candump of this issue. One thing to note, it only happens when I start the terminal after starting the application. 
[candump-2023-04-25_180654.log](https://github.com/ad3154/Isobus-plus-plus/files/11324551/candump-2023-04-25_180654.log)

GwnDaan, 2023-04-27:
Unfortunately that doesn't make a difference :(

[candump-2023-04-27_103439.log](https://github.com/ad3154/Isobus-plus-plus/files/11341434/candump-2023-04-27_103439.log)

GwnDaan, 2023-12-24:
I wasn't able to reproduce this problem with the same terminal and socketcan, so I assume it's a hardware issue. I'll reopen if this issue appears again

closed 2023-12-24 by GwnDaan