Open-Agriculture/AgIsoStack-plus-plus #263: Create documentation for PlatformIO with ESP-IDF

GwnDaan, 2023-05-24:
We get more and more questions about the use of this stack in combination with an ESP (e.g. ESP32). It is known to work with PlatformIO and ESP-IDF, but it is missing some documentation. Therefore it could be useful to have a tutorial page or two dedicated to this. 

- [ ] Documentation on how to use PlatformIO with this stack
- [ ] Example that runs on any ESP with PlatformIO, probably an adapted VT object pool example?

ad3154, 2023-09-10:
@GwnDaan I have a fair bit of this documentation generated now, but I'm sort of learning PlatformIO as I go... do you have a working CMakeLists and platformio.ini file you could share with me, even if they are specific to your application? I'm having a tricky time figuring out the syntax to pull in the TWAI plugin specifically via the CAN_DRIVER CMake variable

GwnDaan, 2023-09-12:
> @GwnDaan I have a fair bit of this documentation generated now, but I'm sort of learning PlatformIO as I go... do you have a working CMakeLists and platformio.ini file you could share with me, even [...]

The TWAI driver should be automatically compiled in by platformio iirc. The cmake variables don't work on platformio as it uses its own compiling. Have a look at the library.json file in our repo which does the file filtering. I'll send the files in a bit

GwnDaan, 2023-09-12:
The `CMakeLists.txt` (basically the standard):
```cmake
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(plough-isobus-control)
```

And a `platformio.ini`:
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[env:debug]
platform = espressif32
board = esp32-s3-custom
framework = espidf

build_type = debug
upload_protocol = esptool
debug_tool = esp-builtin
board_build.embed_txtfiles = src/vtpooldata.iop

monitor_dtr = 0
monitor_speed = 115200
monitor_filters = esp32_exception_decoder

lib_deps = https://github.com/ad3154/Isobus-plus-plus.git
```
that's basically it. And then you can do the includes just fine
```c++
#include "isobus/hardware_integration/can_hardware_interface.hpp"
#include "isobus/hardware_integration/twai_plugin.hpp"
```

ad3154, 2023-09-14:
Amazingly, one of my big issues was that PlatformIO gave me `main.c` instead of `main.cpp` by default... ugh 😣I feel silly for not figuring that out sooner...

AndreaInverardi, 2023-09-16:
Built "esp32_platformio_object_pool" example, as in the tutorial. Deployed on some ESP32 module. I can see some CAN frames been sent at boot. If I check serial port of ESP32 module it says every 5 seconds:

"E (391530) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:"

Tried also connected to some tractor's virtual terminal. It do not detect it.

PR #326 (2023-10-07): Add ESP32 + PlatformIO Tutorial and Example
## What's new

* Added tutorial page for using our stack on ESP32 with PlatformIO. 
* Added ESP32 + PIO example files.

I would like some feedback on this new tutorial before merging it, ideally, so you can find a draft of it [here](https://delgrossoengineering.com/isobus-tutorial/Tutorials/ESP32%20PlatformIO)

Any and all comments would be welcome.

Closes #263
closed 2023-10-07 by ad3154