Open-Agriculture/AgIsoStack-plus-plus #264: Support Arduino IDE

GwnDaan, 2023-05-24:
There is some demand for being able to program a microcontroller using Arduino IDE. I think it could be a good way to get started quickly with this stack. Though I must say I don't have much experience using it.

- [ ] Verify full functionality through Arduino IDE
- [ ] Write documentation on how to get started
- [ ] Adapt an example for 1:1 use with a microcontroller (e.g. ESP32/Teensy) running Arduino Core.

When we commit to support the Arduino IDE, there will be yet another possible way a commit can break something. And since we are constantly making changes at the moment, it could really slowdown progress. Hence waiting with this until a stable version is released may be something to take into considiration.

ad3154, 2023-05-24:
I have actually taken a stab at this in the past and it requires a lot of organization changes (folders and whatnot), plus no `std::thread` etc. So this will almost certainly have to be a separate repo I would say 😅. I think it's doable to make and maintain it separately without too much hassle once we hit "stable" and move to the organization account.

ad3154, 2023-05-30:
It might not have been `std::thread` but it was something threading related, that... maybe `condition_variable`? It's been a while since I tried so it's worth another look for sure. At the very least we may be able to script a patching system and deployment to another repo automatically that follows the Arduino format or something? I'm sure we can work something out to minimize our maintenance effort.

ad3154, 2023-05-30:
I think it will be tough not to have some other repo in some form, as their library manager is mostly just a collection of git repos that it clones in: https://github.com/arduino/library-registry/blob/main/repositories.txt

tomstix, 2023-06-17:
I'd say the biggest problem with general Arduino support is that it is still highly platform dependent on what is available on the C++-side. The Arduino core only defines the general program structure (setup()/loop()) and peripherals like UART/I2C/SPI (but no CAN for example, so this will again be platform dependent). Also not every Arduino implementation supports all of the C++ standard library.
To support std::thread and so on, you'd need an RTOS which not all microcontrollers that can run Arduino have proper support for (I tried with a FreeRTOS port for Teensy 4.1 but didn't have success yet).
However, as the ESP32 is already supported by this library this would be a good starting point to get working with the Arduino IDE. And as the ESP-Arduino core just runs on top of ESP-IDF, this should not be a lot of work.
But to get general Arduino support for most platforms, major rewrites and restructurization of the code is certainly necessary.

aeonSolutions, 2023-06-17:
I maintain a set of ESP32 libraries for Arduino Studio. Have a look here to see if it helps you.
[https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries#readme](https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries#readme)

closed 2023-09-10 by ad3154