Open-Agriculture/AgIsoStack-plus-plus #319: Support for zephyr

GwnDaan, 2023-09-06:
It would be nice to be able to integrate AgIsoStack++ in a zephyr based project. It probably can be done with a link to a git submodule in the cmake. But it may as well be more difficult, like for Arduino #264. Further research on this is needed

tomstix, 2023-11-02:
I was thinking about that as well a while ago. The main problem I saw back then was the lack of std::thread in zephyr. But afaik, support for that is on the way. Or we might just work around that using native zephyr threads / wrappers or something.
What are your findings so far @GwnDaan?

ad3154, 2023-11-02:
We do now support single-threaded operation as well, which could be an option in the meantime if they are working on std::thread. It's what we do on Arduino at the moment

GwnDaan, 2023-11-03:
> What are your findings so far @GwnDaan?

To be honest, I have not looked into this yet. I initially was planning on integrating the stack in an existing zephyr project, but never got to it. So if anyone wants to look into this before that, feel free to do so 😄

ad3154, 2024-03-31:
#458 Adds partial Zephyr support, see [here](https://docs.zephyrproject.org/latest/services/portability/cmsis_rtos_v2.html)

We'll just have to make some zephyr specific tweaks when we detect their OS specific macro in the cmsis_os header path.