Open-Agriculture/AgIsoStack-plus-plus #350: Task Controller Server

Eric-Bwr, 2023-11-21:
Good day,

is there something planned in this regard?
Nice work btw!

GwnDaan, 2023-11-21:
Hey there, welcome to the community! Ultimately we would want to support everything inside the ISO11783 standard with this stack, including the task controller server. Though the task controller server has not yet been touched upon, my wild guess is that it will still take quite a while for us to get to. But maybe @ad3154 has some plans already? Also we welcome any contributions you can make to help drive progress 😄

Eric-Bwr, 2023-11-21:
Hello,
unfortunately, I dont have much experience with the behind the scenes stuff of a tc server. We are using a commercial library which is... lacking some features... And the developers are.... lets say they are too busy or whatever... 
So this library is an absolut dream. But I fear due to my limited time and lack of deep knowledge I cant help you guys out. 
Im more than happy to help in testing though. Ive got several implements at my disposal.

Eric-Bwr, 2023-11-21:
Wonderfull! Sure! Tomorrow I will provide a list of functionality which we have right now, and would still be nice to have, and a list which we are missing out on. Thank you very much!
And of cause no pressure. But lets get this party started :)

Eric-Bwr, 2023-11-22:
Too much work today, sry guys, next week :)

Eric-Bwr, 2023-11-27:
So, lets see what would be nice to have in regards to the TC-Server:

- A way to read in TASKDATA.XML Files and set the store dir path for the coverage
- Setting the task status (Start, Paused, Stop)
- Callbacks for task status 
- Callbacks for AddDevice / RemoveDevice (when Implement connects to can, could handle things like pivot type (trailed or fixed), and getting the implement (booms (section)) geometry aswell as adding a x and y offset to them)
- Callbacks and setter within those for manipulating the section control state 
- Same thing for rate control
- Possibility to set the pre on off time
- Setter for section overlap
- Setter for Products (rate control related)
- Setter to apply general gps offset (not identical to the AddDevice Boom Offset)
- Being able to clear the coverage and set custom values at X and Y index

The lib we use has its own direction api, so they try to compute the working direction based on position data.
We would like to use our own heading information to resolve some latency issues.
But here are some features in that regards:
- Callbacks from direction api (optional -> gps loss, gps update)
- Setter / Getter for machine direction
- Setter / Getter for heading information (or even parse the NMEA2000 cog & sog pgn and other various protocols)

So yeah, this is the ruff idea I guess.
The first steps would be to parse a TASKDATA.XML File and seeing if a implement can connect to the tc server and get its geometry.

ad3154, 2024-05-08:
Closing this issue for now, since we merged it in #407 but we can refer back here for further ideas on how to streamline this interface based on Eric's suggestions

closed 2024-05-08 by ad3154