Open-Agriculture/AgIsoStack-plus-plus #357: CXX_STANDARD 14 Problems in Relation to esp-idf-v5.1.1

franz-ms-muc, 2023-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

**Supporting Documentation**
If your issue is CAN behavior related, it may be helpful to have a CAN trace of your issue that clearly shows the undesirable behavior. Please attach one if you think it might help, or if requested. Try to capture address claiming in addition to the bad behavior if possible. We prefer files captured with `candump -l` or `.asc` ascii format log files.  Binary Vector files `.blf` will not be accepted. 

**Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Compiler [e.g. GCC 9.4.0]
 - CAN Driver [e.g. Socket CAN]

**Additional context**
Add any other context about the problem here that you think might help us recreate or otherwise address your issue if applicable.

GwnDaan, 2023-11-25:
Hi! I'm trying to reproduce this issue, would you be able to provide the full command used to build the project? In particular, where did you find `-std=gnu++2b -fno-exceptions -fno-rtti -std=c++14` arguments?

GwnDaan, 2023-11-25:
@franz-ms-muc I can confirm the build fails when running with the build argument `-std=c++14` for ESP-IDF. Try running the build command only with `-std=gnu++2b`, that works for me.

franz-ms-muc, 2023-11-25:
> @franz-ms-muc I can confirm the build fails when running with the build argument `-std=c++14` for ESP-IDF v5.1.1. Try running the build command only with `-std=gnu++2b`, that works for me.

where do you set these ?

franz-ms-muc, 2023-11-25:
> Hi! I'm trying to reproduce this issue, would you be able to provide the full command used to build the project? In particular, where did you find `-std=gnu++2b -fno-exceptions -fno-rtti -std=c++14` arguments?

you find it in the Output. for every Source File cmake outputs the Called Compiler line. 

[https://gist.github.com/franz-ms-muc/cef012699935ba0fa8be564821bfc212](https://gist.github.com/franz-ms-muc/cef012699935ba0fa8be564821bfc212)

see here for just this line.

GwnDaan, 2023-11-25:
I'm not 100% sure, i tested in PlatformIO, so it's was a little different there, but I think the following would work for you: 

In your own CMakeLists.txt edit the `CXX_STANDARD` variable, and making sure that is done _after_ AgIsoStack is added. So for example:

```cmake
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

or maybe unsetting it could work aswell:

```cmake
unset(CMAKE_CXX_STANDARD)
unset(CMAKE_CXX_STANDARD_REQUIRED)
```

In your case that probably would be at the very end of the agisostack 'component' CMakeLists.txt you configured, or somewhere in the CMakeLists.txt of your project itself. Let me know if that works :)

GwnDaan, 2023-12-12:
Hi @franz-ms-muc, your original issue should be fixed by https://github.com/Open-Agriculture/AgIsoStack-plus-plus/commit/f690bdda95222b72b6c95abfc7c0f91a2f23e74c. Any chance you would be able to test it with your initial setup?

GwnDaan, 2024-01-30:
Closing due to inactivity, feel free to re-open if issue still present

closed 2024-01-30 by GwnDaan