Open-Agriculture/AgIsoStack-plus-plus #359: sample "VT" not loading on a CCI 1200, but loading fine on a CCI 50

franz-ms-muc, 2023-11-25:
https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/AgIsoStack-plus-plus-sample/tree/v0.1

running on a ESP32S3, 
Compiled with esp-idf 5.1.1

Loading fine on a CCI 50: 
https://gist.github.com/franz-ms-muc/ef21d699eb291d7d60ca62171c06df05

Error on a CCI 1200:

GwnDaan, 2023-11-25:
Hmm, the CCI 1200 just responds with "any other error" for the object pool. So it likely is missing something in the object pool. But I'm not sure what. Can you try the following to provide more information:

- Set the CANStackLogger to `DEBUG` ([here](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/AgIsoStack-plus-plus-sample/blob/a9edfc9f6d11febfe4d123a45652f99e924a165b/sample/main/main.cpp#L80))
- Record a candump for the CCI 1200 display when starting the sample
- Which VT version runs on the CCI 1200 display?

GwnDaan, 2023-11-25:
The NAME's of the devices seem correct (https://runkit.com/embed/48w0m4dhjnqc), so I still think it is something with our test object pool.

Maybe another thing you can try; replace the test pool with a pool that you know works on the CCI 1200, if you have one that is

ad3154, 2023-11-25:
Yeah, a candump (or a Vector log, or trc file, or whatever you have available) would tell us if the pool was transferred with out corruption of some kind at least. That would be good to confirm.

I can fairly confidently say though that the pool's schema (assuming it transferred properly) _should_ be valid - it's a very basic pool, with only VT version 3 objects and options. It has been tested against several very picky terminals, like Vector's VT implementation, Deere 2630, etc

diplfranzhoepfinger, 2023-11-29:
Hi, 
we tested it on a Topcon X30, not loading.

diplfranzhoepfinger, 2023-11-29:
we tested your Pool with the logiBUS System, and same things: 
loading on a CCI50, loading on a CCI200, 
not loading on a CCI 1200, not loading on a Topcon X30. 

so it "seems" to be the Pool only. 

i will try to provide logs ASAP.

diplfranzhoepfinger, 2023-11-29:
i ran your Pool trough the AEF Validator: 

```

C:\git\ms\AgIsoStack-plus-plus-sample\sample\main\object_pool>.\PoolValidationTool\aef_pool_validator.exe --iop1 object_pool.iop --capab ObjectPools\Part6\WSM\Cap.txt
VTPoolHandlerValidator::SetCapabilitiesFromFile: new value for attribute "vt_version" = 3
VTPoolHandlerValidator::SetCapabilitiesFromFile: new value for attribute "designator_area_max_x_dots" = 80
VTPoolHandlerValidator::SetCapabilitiesFromFile: new value for attribute "designator_area_max_y_dots" = 80
VTPoolHandlerValidator::SetCapabilitiesFromFile: new value for attribute "number_virtual_softkeys_max" = 10
use default limits specified in 'IS Version ISO11783-6:2010(E), Second Edition'
VTPoolHandler::ParsePool BEGIN
VTPoolHandler::ParsePool: parse object pool...
VTPoolHandler::ParsePool END
VTPoolHandler::WriteErrorsToFile: no errors, object pool is valid! :-)

C:\git\ms\AgIsoStack-plus-plus-sample\sample\main\object_pool>pause
Drücken Sie eine beliebige Taste . . .

```

GwnDaan, 2024-01-30:
[image]

Both of our VT examples seem to be working nicely on my CCI1200, even cozy together hahah. Not sure what the issue was for you, but I expect something might have gone wrong somewhere in the setup. 

Feel free to re-open the issue if the problem you encountered is still present.

hkaze, 2024-09-03:
Hello, sorry for commenting in a closed issue.
I would like to share my findings and solution.

I had similar problems with a `Topcon X23`.
In my setup I used the example:
https://github.com/Open-Agriculture/AgIsoStack-plus-plus/tree/c44ce9691f2b56e889dc891b0046353cb32d2f63/examples/virtual_terminal/esp32_platformio_object_pool
I tested with an ESP32: similar error as OP.
I tested with an ESP32-S3: similar error as OP.

I then used the tool:
https://github.com/moehman/PoolEditDist
To construct my own object pool, to be able to investigate further.

After several days of debugging, the results were:
- The error always happened at the last object of the object pool.
- I then excluded the last object from the pool.
- The error then happened at the "new" last object of the object pool.
- Then I investigated the API function that sends the object pool.

### The solution (in my case) ###

At this line:
https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/c44ce9691f2b56e889dc891b0046353cb32d2f63/examples/virtual_terminal/esp32_platformio_object_pool/src/main.cpp#L138
- Change `-1` to `+1`.

In the OP example, at this line:
https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/AgIsoStack-plus-plus-sample/blob/a9edfc9f6d11febfe4d123a45652f99e924a165b/sample/main/main.cpp#L111
- Change `-1` to `+1`.

My reasoning:

Imagine this "memory table" with the object pool:
| ADDRESS  | OBJECT POOL |
| ------------- | ------------- |
| 123  | object_pool_start  |
| 124  | object_pool_inside  |
| 125  | object_pool_inside  |
| 126  | object_pool_inside  |
| 127  | object_pool_end  |

We can certainly see in this fictional example that the size is `5` positions.
Then the math is: `object_pool_end - object_pool_start + 1` : 127-123+1 = 5

It works after applying this correction to the size sent with the API `virtualTerminalClient->set_object_pool`.
Then I could send the object pool from the original example, without modification to the object pool.

diplfranzhoepfinger, 2024-09-03:
@hkaze i did a corresponding commit. 
but not yet tested. 
will test later.

hkaze, 2024-11-12:
Hello, I have an update.
My coworker tested my reasoning with a Topcon VT and an Agres VT (two brands).
My previous reasoning is WRONG. Sorry.
It did not work with Agres display:
[image]

The `object_pool_end` is NOT inside the object pool.
i.e. the pointer of "object_poll_end" does not point to the last byte of the object pool.

From the `.map` file of the `build` folder of ESP32-S3 (my object pool; not the example):
[image]

We have:
- `_binary_object_pool_iop_start = 0x3c100120`
- `_binary_object_pool_iop_end = 0x3c103e7f`

And:
[image]

We can see that the real size is: `15711 bytes` (my object pool; not the example).
That is equal to `0x3c103e7f - 0x3c100120`.

Therefore, the correct size is: `_binary_object_pool_iop_end - _binary_object_pool_iop_start`.

The correct should be:
`virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start), "ais1");`
- Without the `+1`.

By the way, with Topcon display, both worked:
`virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start) + 1, "ais1");`
and
`virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start), "ais1");`

With Agres display, only the correct size worked:
`virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start), "ais1");`

## TL;DR ##

`virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start), "ais1");`

closed 2025-01-08 by GwnDaan