Open-Agriculture/AgIsoStack-plus-plus #417: Expand namespacing with differentiating components

GwnDaan, 2024-01-25:
I've stumbled upon constants being hidden inside declarations of classes for some time now. I find that it makes it less appealing to use them, because it adds unnecessary inheritance in the class wanting to use them. I know we have the `can_constants.hpp` header file and I would like to use it, however throwing everything on one pile can be less appealing as there are a few domain-specific constants that will bloat the isobus namespace universally. 

So a few proposals, which I very much like feedback on:
- Rename the "isobus" namespace to "agisostack" - sounds like a big deal
- Add namespaces "virtual_terminal", "implement_messages", "tractor_ecu", "task_controller", "diagnostics", "file_server" and "sequency_control"
- Add the classes that fit into any of these namespaces to it and remove their respective prefixes
  E.g. `isobus::VirtualTerminalClient` -> `isobus::virtual_terminal::Client`, etc.

If anyone has any thoughts on this, please let me know :)

ad3154, 2024-01-25:
Oof, I dunno... besides the potential of obliterating all consuming applications compilation (which seems undesirable) the main benefit of shorter names has never been a particularly compelling one to me, as you lose the local context of which `Client::` or whatever you are looking at when maintaining or reading the library's components. Plus, in ISO11783 there's lots of overlap between things like VT and TC and Implement messages... I guess I'd be happier trying to figure out better solutions to specific constants you're having pains with

GwnDaan, 2024-01-25:
Okay, that's indeed undesirable for resolving a small issue. So what I would need now is the `NULL_OBJECT_ID`, which indeed is overlapping between TC / VT so the namespaces wouldn't have made much sense. Perhaps just adding it to the constants header file is fine?

closed 2024-01-25 by GwnDaan