Open-Agriculture/AgIsoStack-plus-plus #428: advanced-example / SeederExample e -  only 1 of 6 sections is controlles by TC-Client

engel2, 2024-02-05:
**Describe the bug**
The SeederExample shows six supported sections in the task controller. But there is only one section switched. The TaskControllerClientExample works fine with 16 sections and the same hardware setup (same TC-Server).

**Supporting Documentation**
[SeederExample-OneSectionOnly.log](https://github.com/Open-Agriculture/AgIsoStack-plus-plus/files/14164389/SeederExample-OneSectionOnly.log)

**Environment (please complete the following information):**

- OS: Win10
- Compiler GCC 12.2.0
- CAN Driver PCAN-USB
- Müller Touch 1200 with ISOBUS-TC
- NMEA0183 Simulator

**Additional context**
Add any other context about the problem here that you think might help us recreate or otherwise address your issue if applicable.

https://github.com/Open-Agriculture/AgIsoStack-plus-plus/assets/50421355/544957d6-9d58-435c-a9d7-9301daf779e2

[image]

GwnDaan, 2024-02-05:
Hi there, which branch are you testing the seeder-example on? Could you try the "daan/advanced-example" branch and see if that's any better?

engel2, 2024-02-05:
I tested it with this branch and I think that I identified the bug.
The break command is on the wrong position. 
I put it in the else statement and all sections are working. Don't know if this is well coded, but it works so far.  Maybe you can check it.
My fixed version:
[image]

version from "daan/advanced-example":
[image]

PR #429 (2024-02-12): [Example] Update seeder example 
## Describe your changes

Update seeder example to use latest changes, including VT state tracker/updater. 

Other changes:
- Allow settings auto/manual mode from TC server.
- Dropped the (actual) section states and instead use either the setpoint or switch state to determine the actual state of a section.
- Reworked alarms to be more intuitive to maintain and made it more robust 

Fixes #428 

## How has this been tested?

Ran the seeder example on several terminals
closed 2024-02-12 by GwnDaan