Open-Agriculture/AgIsoStack-plus-plus #446: SC: Pre On Off Time

Eric-Bwr, 2024-03-08:
Hello,

is there any possibility to change the section control pre on off time?
for the tc client and/or server?

Maybe I missed it :)

ad3154, 2024-03-09:
This one is kind of a tricky one that I think has multiple approaches.

A few options come to mind for how one could do to do this "right".

1. Read (or send) the DDIs for Physical Setpoint/Actual Time Latency (DDIs 142 and 143)

```
The Setpoint Value Latency Time is the time lapse between the moment of receival of a setpoint value command by the working set and the moment this setpoint value is physically applied on the device. That means if the setpoint value is communicated on the network (CAN bus) but the system needs 2 seconds to adjust the value physically on the desired unit (device element) then the Setpoint Latency Time is 2 seconds. 
```

2. Implement a proprietary DDI for this on both sides, which specifies the look-ahead time for section control
3. Add an application specific way to set the look-ahead time for section control, in which case you'd just delay when you send the commanded setpoint work state for a given section.

 The way that our TC  server and client are written it kind of leaves this sort of functionality up to the application to interpret based on the vehicle speed and what you send or receive in the DDOP...

I am not really sure where to go with it architecturally in our stack, as those are all kind of application specific things. Do you have any thoughts about what it could look like @Eric-Bwr  ?

gunicsba, 2026-11-05:
The way this works on a Müller Sprayer is like this:
When we have any section on it uses SC Pre On/OFf time of 0.8 and 0.4 seconds respectively.
When we have all the sections off we'll have to open the master valve, adjust the pressure etc so we'll need a bigger value. In this case the implement asks the TC to send Section informations sooner so it increases the Pre On Time.

I think your application is similar to this right @Eric-Bwr  ?