Open-Agriculture/AgIsoStack-plus-plus #449: Error occurred in seeder_example

JayQu9, 2024-03-13:
**Describe the bug**

After loading the object pool, a quick click on the settings button gives me an error
[Info][VT]: Loaded object pool version from VT non-volatile memory with no errors.
[Warn][VT]: Server response to a command timed out
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 18446744073674967472)

Sometimes, an error occurs after the Server response to a command timed out

[Warn][VT]: Server response to a command timed out
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

**Environment (please complete the following information):**
- OS: win10
- Compiler GCC 7.3.0
- CAN Driver PCAN

GwnDaan, 2024-03-13:
Hi there, thanks for reporting this issue.

Since there is no trace in the error message currently, it is hard for us to find the exact location. Can you run a debug session so it stops at the point where the exception occurs, and provide us with the stack trace?

JayQu9, 2024-03-13:
After clicking debug, continue to click the setting soft key immediately before the object pool is loaded, it will exit directly.
[image]
[image]

closed 2024-04-22 by JayQu9