Open-Agriculture/AgIsoStack-plus-plus #468: 32bit identifier messages Linux

preverte, 2024-05-15:
Hi all, 
Reviewing ISOBUS concepts, as mentioned in the [concepts section](https://agisostack-plus-plus.readthedocs.io/en/latest/Concepts.html), identifiers are **32 bit** long. 
In Linux, however, when trying to send a 32bit-id message through _cansend_, the messages I receive using _candump_ have been masked to 29bits:
[image]
I was wondering if anyone has some experience with 32bit identifier transmission with SocketCAN and, also, if through this library it is possible to send and receive this type of messages. If so, which tool could I use for message transmission debugging? 

Thank you in advance!

ad3154, 2024-05-15:
Physically the identifier is 32 bits, and we store it in a u32, but in reality, ISO11783 and J1939 standard all define it as 29 useable bits. Therefore, sending messages with non-compliant IDs is out of scope for this project, and shouldn't be done on such a network.

The three missing bits are "RTR", "SRR", and "IDE" and are controlled entirely by the CAN specification. They exist to allow backwards compatibility with CAN2.0a and are invisible to all OSI layers above the data link layer, just like the ACK, SOF, and EOF parts of a frame.

As for `cansend`, I believe it lets you set only the RTR bit out of those 3 bits.

See [their documentation](https://manpages.debian.org/testing/can-utils/cansend.1.en.html).

However, ISO11783-3 states, "Remote Transmission Request (RTR = 1) is not available for use in the ISO 11783 network."

So, my suggestion would be to keep it to 29 bits, since those are the useable bits.

preverte, 2024-05-15:
Thanks for the response. That was my main understanding... 29 "useful" bits that compose the identifier and 3 more bits for communication purposes. 
As for now I was able to send and receive compilant IDs successfully with the library, kudos!

closed 2024-05-15 by ad3154