Open-Agriculture/AgIsoStack-plus-plus #476: Loading object pool to GS3 2630 display failing

ironstone-ag, 2024-06-15:
Hi everyone

Very new to c++ so I appreciate your patience with the next few silly issue posts.

OS: Windows 11 10.0.22631
Compiler: Visual Studio 2022 with CMake tool
CAN driver: PCAN-USB

The bug: when running the SeederExample and VT3Example with a Topcon X35 everything works perfectly. But trying to use a John Deere 2630 display I keep running into a log error: 

[Error][VT]: Error in end of object pool message.Faulty Object 65535 Faulty Object Parent 65535 Pool error bitmask value 15

So it seems the issue is the bitmask value 15 but I just get to a dead end in the code and don't know what 15 means.

Additional context/red herring:
I have Jetter ISO-Designer and imported the VT3Example object pool, removed the image files, and using Version 3 setting exported the new .iop file to the example. It will upload successfully to the 2630, buttons and strings work correctly, but then if i add in more output strings once I reach a total of 22 strings I get the same error message previously described. Deleting one output string makes it work again. I may be going down a very wrong path here but it's the only way so far I can get the object pool to load successfully.

The 2630 has an option for ISOBUS certification mode. Turning on doesn't seem to make a difference.

Log:
Loaded object pool from VT3TestPool.iop
[Info][NM]: Partnered control function with name a0001d00042708d7 has claimed address 38 on channel 0.
[Info][VT]: No version label from the VT matched. Client will upload the pool and store it instead.
[Warn][VT]: Get Versions Response ignored!
[Error][VT]: Error in end of object pool message.Faulty Object 65535 Faulty Object Parent 65535 Pool error bitmask value 15
[Info][VT]: Resetting Failed VT Connection
[Info][VT]: No version label from the VT matched. Client will upload the pool and store it instead.
[Error][VT]: Error in end of object pool message.Faulty Object 65535 Faulty Object Parent 65535 Pool error bitmask value 15
[Info][VT]: Resetting Failed VT Connection
[Info][VT]: No version label from the VT matched. Client will upload the pool and store it instead.

I would like to take a CAN trace, if anyone can please let me know how with my hardware/system that would be great.

Thank you

ad3154, 2024-06-15:
Hey there, I'd be happy to help you if I can.

One important thing about the 2630 is that it's VT version 3 and **_very_** strict about that, so if you have any version 4+ options set on your output strings (justification settings of certain types for example) it will reject it.

If you feel comfortable attaching the IOP file or a CAN trace, I'd be happy to check it over for you and see what's going on.

If you'd rather not attach it here, you can also drop it in our telegram group, our discord group, or DM me on one of those platforms with it.

ironstone-ag, 2024-06-16:
The unmodified .iop files included in VT3Example and SeederExample both cause the error. I'm happy to upload a new .iop file here as all i've done is delete images to make it work, although interestingly when images are deleted it no longer works on Topcon X35.

What's the best way for me to record a CAN trace? I have P-CAN View which normally can record traces but I can't access the P-CAN USB device when example program is running unfortunately, it won't let 2 programs simultaneously access the hardware.

Attached are two .iop files, the orginal one supplied with VT3Example, and one I've removed the 4 images from. As I deleted the image from the working set which requires at least one object, I have just added a simple line accross it.
[VT3TestPool.zip](https://github.com/user-attachments/files/15855800/VT3TestPool.zip)

ad3154, 2024-06-16:
That's strange, as neither one seems to have any issues with VT version 3 from what I am seeing.... another thing could be that deere displays seem to expect a fixed maximum number of ETP packets per EDPO, so another thing you could try is setting that to 255.

`isobus::CANNetworkManager::CANNetwork.get_configuration().set_number_of_packets_per_dpo_message(255);`

This could help...

As far as taking a CAN trace, since you are running on a PC, I think I can probably hook you up with a plugin for our stack to log all the messages at the same time as the normal stuff it does. I had to write something like that once for our VT as well, so I will see if I can get you something - will get back to you on that.

ad3154, 2024-06-16:
> What's the best way for me to record a CAN trace? I have P-CAN View which normally can record traces but I can't access the P-CAN USB device when example program is running unfortunately, it won't let 2 programs simultaneously access the hardware.

I've added some native CAN trace functionality to the stack to help you and others with things like this.

https://github.com/Open-Agriculture/AgIsoStack-plus-plus/pull/477

Check out the class `VectorASCLogger`. If you create one, it will allow you to both take a CAN trace, and run the application at the same time, without needing another CAN adapter.

ironstone-ag, 2024-06-17:
Sounds great, thanks for responding so fast! I've now added to main.cpp #include "isobus/hardware_integration/vector_asc_logger.hpp" and called isobus::VectorASCLogger::VectorASCLogger(); as the first line of my main(). It's generating an .asc log file each launch but only has the following contents:

date date Tue Jun 17 11:8:46 am 2024
base hex timestamps absolute
no internal events logged

What have I missed here? Is there a proper way to 'exit gracefully' instead of just closing the window?

ad3154, 2024-06-17:
Well, that will call the constructor but will immediately destruct the object, haha.

Try something like this instead:
[image]

The difference being that we create a local variable on the stack rather than just calling the constructor.

The example supports exiting by doing ctrl+c which is a more graceful than closing the window

ironstone-ag, 2024-06-17:
It had to be something simple! Thanks for explaining. Here are some logs and accompanying .iop files from running VT3 example with 2630 with differences:

1. completely unmodified, fails to upload
[1. unmodified.zip](https://github.com/user-attachments/files/15862186/1.unmodified.zip)

2. modified .iop to remove 4 image files, add in 28 output strings, successfully uploads
[2. modified iop no images 28 string variables.zip](https://github.com/user-attachments/files/15862188/2.modified.iop.no.images.28.string.variables.zip)

3. same as 2. but added just one more output string and fails to upload
[3. modified iop no images 29 string variables.zip](https://github.com/user-attachments/files/15862190/3.modified.iop.no.images.29.string.variables.zip)

4. same as 1. but packets per DPO message set to 255 as instructed, fails to upload
[4. unmodified iop and packet per DPO set to 255.zip](https://github.com/user-attachments/files/15862191/4.unmodified.iop.and.packet.per.DPO.set.to.255.zip)

ad3154, 2024-06-17:
Well, the reason it seems to be failing is not related to the pool content, but rather something it is doing (incorrectly I might add) with ETP, as it starts failing as soon as it needs to receive ETP from us, and works when it receives TP. We have seen some bad behavior from them and ETP before... ugh.

So getting the packets per DPO thing working seems important to fixing it, as that has helped before on their Gen4 displays, but
in your log number 4, it looks like the stack is only sending out 39 frames of ETP, and then stops? Which is very strange.... so we'll have to get to the bottom of that I think to get it working for you.

Adding @GwnDaan in case he has any ideas as to why that could be happening, but I will try and set up some way to test this again on my end as well...

ad3154, 2024-06-17:
Could I ask, where did you put the code to change the DPO packet count in main.cpp? Could you attach the cpp file here with the modification made? I want to be sure I am recreating it exactly

ironstone-ag, 2024-06-17:
Sure thing, here is main.cpp I used for log file 4

`#include "isobus/hardware_integration/available_can_drivers.hpp"
#include "isobus/hardware_integration/can_hardware_interface.hpp"
#include "isobus/isobus/can_network_manager.hpp"
#include "isobus/isobus/can_partnered_control_function.hpp"
#include "isobus/isobus/can_stack_logger.hpp"
#include "isobus/isobus/isobus_virtual_terminal_client.hpp"
#include "isobus/isobus/isobus_virtual_terminal_client_update_helper.hpp"
#include "isobus/utility/iop_file_interface.hpp"

#include "isobus/hardware_integration/vector_asc_logger.hpp"
#include "console_logger.cpp"
#include "objectPoolObjects.h"

#include <atomic>
#include <csignal>
#include <iostream>

//! It is discouraged to use global variables, but it is done here for simplicity.
static std::shared_ptr<isobus::VirtualTerminalClient> virtualTerminalClient = nullptr;
static std::shared_ptr<isobus::VirtualTerminalClientUpdateHelper> virtualTerminalUpdateHelper = nullptr;
static std::atomic_bool running = { true };

void signal_handler(int)
{
	running = false;
}

// This callback will provide us with event driven notifications of softkey presses from the stack
void handle_softkey_event(const isobus::VirtualTerminalClient::VTKeyEvent &event)
{
	if (event.keyNumber == 0)
	{
		// We have the alarm ACK code, so if we have an active alarm, acknowledge it by going back to the main runscreen
		virtualTerminalUpdateHelper->set_active_data_or_alarm_mask(example_WorkingSet, mainRunscreen_DataMask);
	}

	switch (event.keyEvent)
	{
		case isobus::VirtualTerminalClient::KeyActivationCode::ButtonUnlatchedOrReleased:
		{
			switch (event.objectID)
			{
				case alarm_SoftKey:
				{
					virtualTerminalUpdateHelper->set_active_data_or_alarm_mask(example_WorkingSet, example_AlarmMask);
				}
				break;

				case acknowledgeAlarm_SoftKey:
				{
					virtualTerminalUpdateHelper->set_active_data_or_alarm_mask(example_WorkingSet, mainRunscreen_DataMask);
				}
				break;

				default:
					break;
			}
		}
		break;

		default:
			break;
	}
}

// This callback will provide us with event driven notifications of button presses from the stack
void handle_button_event(const isobus::VirtualTerminalClient::VTKeyEvent &event)
{
[... 95 lines cut ...]
	const std::vector<isobus::NAMEFilter> vtNameFilters = { filterVirtualTerminal };
	auto TestInternalECU = isobus::CANNetworkManager::CANNetwork.create_internal_control_function(TestDeviceNAME, 0);
	auto TestPartnerVT = isobus::CANNetworkManager::CANNetwork.create_partnered_control_function(0, vtNameFilters);

	virtualTerminalClient = std::make_shared<isobus::VirtualTerminalClient>(TestPartnerVT, TestInternalECU);
	virtualTerminalClient->set_object_pool(0, testPool.data(), testPool.size(), objectPoolHash);
	virtualTerminalClient->get_vt_soft_key_event_dispatcher().add_listener(handle_softkey_event);
	virtualTerminalClient->get_vt_button_event_dispatcher().add_listener(handle_button_event);
	virtualTerminalClient->initialize(true);

	virtualTerminalUpdateHelper = std::make_shared<isobus::VirtualTerminalClientUpdateHelper>(virtualTerminalClient);
	virtualTerminalUpdateHelper->add_tracked_numeric_value(ButtonExampleNumber_VarNum, 214748364); // In the object pool the output number has an offset of -214748364 so we use this to represent 0.
	virtualTerminalUpdateHelper->initialize();

	while (running)
	{
		// CAN stack runs in other threads. Do nothing forever.
		std::this_thread::sleep_for(std::chrono::milliseconds(1000));
	}

	virtualTerminalClient->terminate();
	isobus::CANHardwareInterface::stop();
	return 0;
}
`

ironstone-ag, 2024-06-17:
Sure thing, here is main.cpp I used for log file 4
[main.zip](https://github.com/user-attachments/files/15878155/main.zip)

ad3154, 2024-06-18:
OK, well, I was able to recreate that specific issue at least with AgIsoVirtualTerminal (our VT).

Seems to be happening because we're not checking the return value of the push into the hardware layer queue, sorry about that.
[image]

Opened PR #479 to fix this. I would bet that if you switch to that branch or wait till we merge it into `main`, that it might work on your 2630 with the number of frames per DPO set to 255 and this change.

This ended up being a pretty big bug, so I really appreciate that you took the time to open an issue with us - thanks!

ironstone-ag, 2024-06-18:
Great success, can confirm working on 2630 with standard VT3Example and SeederExample, plus my modified .iop files in VT3Example. Thanks a million for finding it!

PR #479 (2024-06-18): [HW]: Fixed possibly dropping messages on Tx when Tx buffer is full
## Describe your changes

Fixed an issue where we weren't checking the return value of the push into our Tx queue in our hardware interface. This could have caused some frames to be accidentally dropped.

## How has this been tested?

Recreated the issue discussed [here](https://github.com/Open-Agriculture/AgIsoStack-plus-plus/issues/476#issuecomment-2174568075) with AgIsoVirtualTerminal with the number of frames per DOP set to 255, and witnessed frames being dropped.

Added this check, and now object pool properly uploads!

T [... cut]
closed 2024-06-18 by ad3154