Open-Agriculture/AgIsoStack-plus-plus #480: ESP32-WROOM-32 Can't upload

bartdereu, 2024-07-02:
Hi all,

I've followed the guidelines to open a project in PlatformIO and the build was succesfull , but
i'm having issues with my ESP32-DEVKitC , i can't seem to find a solid way to upload to it.

In the Arduino enviroment, i could upload to it after i added a 10mF capacitor between EN and GND. 
But in PlatformIO, it tries to connect to my comport without luck. 

I've tried :
- Pushing the bootloader button on upload and release it while it says 'Connecting'.
- added the COM port in the platformio.ini file ( upload_port = COM4  in my case)
- also tried  \.\COM4
- tried the other buttons on the ESP32 in all different combinations.

i'm at the end of this and start to think the ESP is defect .could this be?

Matthieu-Poulain, 2024-07-03:
Hi ,

Did you try pushing button 'reset', then 'boot' then release reset and after 'boot' in that exact sequence order?

Note that if you are using platformIO, you must not use the IDF shortcuts but platformIO ones;

A few months ago i started using it with plateform.io, but i decided to switch to a project using only esp IDF and visual studio code Espressif extension because it did not support the lattest IDF versions.

Everything works just fine, except that the example gave the following error:

VirtualTerminalClientInterface->set_object_pool(0, ObjectPool, **(object_pool_end - object_pool_start) **-1** , "V1.X.1"); 
but i had to switch it to 
VirtualTerminalClientInterface->set_object_pool(0, ObjectPool, **(object_pool_end - object_pool_start)** , "V1.X.1"); 
as it cause and error when uploading my custom ObjectPool.
`

bartdereu, 2024-07-03:
Yes, i have tried that combination , same result :

```
Uploading .pio\build\esp32dev\firmware.bin
esptool.py v4.5.1
Serial port COM4
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Matthieu-Poulain, 2024-07-04:
Hmm... Did you ever try flashing it using a simple example from the IDF, without platformIO?