Open-Agriculture/AgIsoStack-plus-plus #490: Question about AUX inputs 2

Matthieu-Poulain, 2024-09-19:
Hi guys,

As a customer has requested support for ISOBUS in our system, we are currently investigating whether we can fulfill this request. We have tested several ISOBUS features using AgIsoStack (great work, by the way, many thanks for your effort), including soft buttons and auxiliary inputs.

Everything worked fine on a neighbor's tractor, but when we tried the same test on the client's tractor, we were unable to find the Aux Functions to bind to the tractor joystick—they simply weren't displayed. Initially, we thought it was a mismatch in the Aux Function type (e.g., boolean latching, non-latching, etc.), but none of the different types worked.

The Virtual Terminals (VTs) on both of his Fendt tractors failed to work. My question is, have you ever encountered this behavior where your projects work on some VTs but not on others?

Additionally, I simply declared the Aux Functions in IsoDesigner. Do I need to work with object pointers, or are they unrelated?

Many thanks for any help you can provide!

ad3154, 2024-09-20:
> currently investigating whether we can fulfill this request

Are you using our CAN stack for this in your system? If it's not with our stack I am not sure how much help we can offer, but if it's AgIsoStack we can see what we can find out!

I am not a big expert on Aux-N to be honest, I have been relying on @GwnDaan for much of it 😅 but what would really help us to understand the issue you've been having is to share with us a CAN log showing your system connecting to the VT if possible, along with any logging output from the stack itself

Matthieu-Poulain, 2024-09-21:
Thank you for your answer. I indeed used your stack with an esp32 to begin the project. Shame on me, i did not log the output when i was testing the system. I did try to set the output log in debug mode but it seems i couldn’t catch and kind of error between the flow of open/close session messages. 

I was hoping to see first if any of you knew if the display of any Aux fonction/input was bound to the stack or if given a .iop file the VT was the only part required to parse, detect and submit it to user selection.

2nd and most recent fendt display seemed to group aux fonctions under a tab, so maybe i missed something here…

cerdei, 2024-11-08:
**The primary terminal** (VT number=1 or VT function instance=0) should check and list the AUX-N assignments, so the operator can view and edit them on this terminal only -> check VT number is 1 and no VT number conflict in case of multiple terminals.
On the other hand the ECU with the AUX function objects in its implement object pool should store these assignments in its non-volatile memory and should communicate it to the primary terminal at startup (and at some other special trigger events).
**Auxiliary Object Pointers:** can be used on AUX capable ECU's data masks (for input units like a joystick, or for your ECU with AUX function objects).
The primary terminal should show or list the related actual assigments (AUX function or input objects) when displaying these object pointers.

Matthieu-Poulain, 2024-11-19:
> **The primary terminal** (VT number=1 or VT function instance=0) should check and list the AUX-N assignments, so the operator can view and edit them on this terminal only -> check VT number is 1 and [...]

Hi, thank you for your clarifications. So if i understand well, the stack should report to the VT which AUX-functions are available in the .iop file, and if they are not displayed, it may be because of a faulty report?

Does anyone know where in the code these interactions are handled and where i should dig?

cerdei, 2024-11-20:
Hi, there is no "report" for AUX function objects, these objects are simply part of the iop, and the (primary) terminal should detect them during the iop parsing process.
So that's why you have to declare these AUX function objects in your favorite pool editor (like ISO-Designer/Maskmaker).
You can also declare the AUX object pointer (Auxiliary Control Designator Type 2 Object Pointer) objects and place them on your data mask, but this is optional and not necessary for the basic functionality.
Based on your [first comment](https://github.com/Open-Agriculture/AgIsoStack-plus-plus/issues/490#issue-2537321679) the issue may be on the (Fendt) terminal side, but it is hard to tell without a CAN log.
If you encounter any problem, saving the actual terminal product name/type (like Fendt Varioterminal 10.4" or 7") and the CAN log for further investigation is always a good practice.

Matthieu-Poulain, 2024-11-20:
Thank you so much for these clarifications.
So the VT is indeed responsible for parsing the .iop from where it should extract the Aux functions available for affectation.
Indeed, data loggin will be on my todo list on next test session, I was lacking experience :/ 

I understand i dont need AUX object pointers, but in order to have the full picture in mind, would you say that once added in the data mask, they would act kind of like a button?

cerdei, 2024-11-20:
The AUX object pointers are some kind of visual helpers for the operator on the mask of the implement (baler, sprayer, etc.) or input unit (joystick, arm controller, switch box, etc.), for example:
The operator can use the mask of the joystick and baler to check the device/machine's graphical representation with the related AUX assignments placed close to its input (joystick button) or function (open baler door, stop baler belt, etc.).

An AUX object pointer's content is displayed by the terminal based on the actual (network-wide) AUX assignments, 
so the operator can easily identify which - graphically represented - AUX function or input has:
- no assignment: a terminal-specific "icon" (X or something),
- one assignment: the assigned AUX function or input object's designator,
- multiple assignments: multiple designators or
a terminal-specific icon ("+++" or something) to show them in a "popup window" or (filtered) assignment list (by pressing/activating the icon).

[In this short video](https://www.youtube.com/watch?v=CQVQi9jyDlw) you can see:
- [0:22](https://youtu.be/CQVQi9jyDlw?si=lwGx_YwmcTXQm2i5&t=22): arm controller mask - "no assignment" X icon inside all AUX object pointers
- [0:35](https://youtu.be/CQVQi9jyDlw?si=lwGx_YwmcTXQm2i5&t=35): terminal's (built-in) AUX assignment editor - adding an assignment to the arm controller
- [1:16](https://youtu.be/CQVQi9jyDlw?si=lwGx_YwmcTXQm2i5&t=76): arm controller mask again - the assigned AUX function (designator) is displayed in the related AUX object pointer (which is referencing the arm controller's specific input object)

Extra info: some terminals can be set up for AUX-O or AUX-N functionality,
please check it on the problematic terminal, [like this (Fendt)](https://www.youtube.com/watch?v=xkcX23bMj2Y).

Matthieu-Poulain, 2024-11-21:
Nice I understand it now. Thank you very much for your help!

Matthieu-Poulain, 2025-01-28:
Hi guys,

I found the issue with the fendt terminal.
While implementing something to store the preferred assignments in my esp32, i removed lines that i had no idea would impact the displayed available Aux functions... What i can't explain is that this did work on the first Deere tractor i used to test.

(Note that the send_auxiliary_functions_preferred_assignment() was sent just ahead of this block, in the load_preferred_assignment_from_NVS function...

```
CANStackLogger::CAN_stack_log(CANStackLogger::LoggingLevel::Info, "[HQ_DBG]: LOADING PREF ASSIGNMENTS (Step 0).");									
parentVT->load_preferred_assignment_from_NVS(parentVT); // Does the parentVT->send_auxiliary_functions_preferred_assignment() inside
									
									
// if (parentVT->auxiliaryFunctionEventDispatcher.get_listener_count() > 0)
//{
//	if (parentVT->send_auxiliary_functions_preferred_assignment())
//	{
//		CANStackLogger::CAN_stack_log(CANStackLogger::LoggingLevel::Debug, "[AUX-N]: Sent preferred assignments after LoadVersionCommand.");
//	}
//	else
//	{
//		CANStackLogger::CAN_stack_log(CANStackLogger::LoggingLevel::Warning, "[AUX-N]: Failed to send preferred assignments after LoadVersionCommand.");
//	}
//}
```

Now i still have an issue to load those prefered AUX-N after a reboot.
It works fine when i reboot the esp32 and not the VT. However if i start both VT and ESP32 at the same time and VT is not up when esp32 stack is, then i loose my assignments. Has someone tried to implement AUX-N loading?

I tried to set this up from this commit here made by @GwnDaan :  [0f845b2](https://github.com/GwnDaan/AgIsoStack-plus-plus/commit/0f845b2d587700fa4d8272e5e7d2f16e9be88d0d)

On another note, i found this line to be maybe incorrect:
```
case static_cast<std::uint8_t>(Function::AuxiliaryAssignmentTypeTwoCommand):
{
	if (14 == message.get_data_length())
	{
		std::uint64_t isoName = message.get_uint64_at(1);
		//OLD VALUE:  bool storeAsPreferred = message.get_bool_at(9, 7);
		bool storeAsPreferred = !message.get_bool_at(9, 7);
```

As i found that the value to store as prefered is 0 in the ISO documents:

> Lorsque l’opérateur a décidé d’attribuer un objet Entrée auxiliaire de type 2 à un objet Fonction 
auxiliaire de type 2, le stockage comme bit d’attribution privilégiée doit être réglé sur 0. Maintenant, 
cette attribution doit être stockée comme attribution privilégiée. Dans toutes les autres situations, elle 
doit être réglée à 1

> When the operator has decided to assign an Auxiliary Input type 2 object to an Auxiliary Function type 2 object, storage as the privileged assignment bit must be set to 0. auxiliary function type 2, the storage as privileged allocation bit must be set to 0. Now, this assignment must be stored as a privileged assignment. In all other situations, it must be set to 1

As a result, if we use storeAsPreferred , we must invert the value of the bit...
Hope this can be of help.

GwnDaan, 2025-01-28:
> As a result, if we use storeAsPreferred , we must invert the value of the bit...

Hi @Matthieu-Poulain, you are absolutely right. That needs to be fixed.

The commit you linked has not been tested, it was just a simple draft. Hopefully you will manage to get it working! Without the full code I'm afraid we can't help you much .

It would be greatly appreciated if you could submit a PR with the necessary changes once the auxiliary assignments are fully functional. That would be really great

Matthieu-Poulain, 2025-01-28:
Hi @GwnDaan, 

Sure thing, I'll share with you what i find, however i did not go through the save engine you created, as i had to get it to work fast, i changed a bit the way it worked specifically for my ESP32, but maybe that could be usefull.

Would you happen to have an idea as to where should I place the functions that load the assignments in order to ensure this is done after the complete boot of the VT?
I found 3 places in the persistent-aux-n branch, but i'm yet to have this to work. Maybe it's related to the missing part of code of my previous message.

GwnDaan, 2025-01-28:
I have been digging a bit more into this and found the following:

> Would you happen to have an idea as to where should I place the functions that load the assignments in order to ensure this is done after the complete boot of the VT?

The functions are already sent at two places in the VT client:

1. After the LoadVersionCommand has been received https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/80e95fe68e61ce9b6f33670cca2f700ed53e5d43/isobus/src/isobus_virtual_terminal_client.cpp#L2947-L2958
2. After the EndOfObjectPoolMessage has been received
https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/80e95fe68e61ce9b6f33670cca2f700ed53e5d43/isobus/src/isobus_virtual_terminal_client.cpp#L3071-L3082

This is the correct places for cases where there is no preferred assignment available, or where all the preferred auxiliary input(s) units are ready.

But we should sent the preferred assignment again when:
- a preferred auxiliary input unit has become ready (and previously was not detected or has reinitialized)
- an auxiliary unit with active assignments has gone offline, and/or reconfiguration alternate auxiliary input units is wanted

I hope this helps in debugging the issue. (For more detailed info see chapter J.7.7 of part 6 of the standard)

Matthieu-Poulain, 2025-01-28:
I also looked more in depth at what the code was doing there this afternoon, and i had the chance to read part of the standard (by the well with an awfull french translation...).
I'm glad we came to the same conclusion regarding the "a preferred input unit is ready (and was previously not detected or not ready)" point. 
I had previously spotted the right place for it in the persistent-aux-n branch, but i did not understand as I should have the right behavior awaited. 
As i do understand it now, the AUX-N assignments save should not be reloaded by default while the Inputs are not initialized and ready (where i previously loaded and pushed them anyway at the LoadVersionCommand and EndOfObjectPoolMessage). So i may have to reconsider how i do my load function.

I had not anticipated the 2nd point though, with the online reconfiguration. I might have to look deeper to where those interractions happen.

On another note, an interrogation that i had was about if i should call the send_auxiliary_functions_preferred_assignment every time a "new" preferred input unit is ready or if i should wait for more to come by using something like a timer, triggering the call after a certain delay, which would reset before triggering if another input was ready.

Anyhow, I thank you for your help, it came really handy. 
I should be able to implement those changes and test them on a fendt terminal by next week. I'll let you know how it went and come back to you to see how those changes could be integrated in the stack.

GwnDaan, 2025-01-31:
> On another note, an interrogation that i had was about if i should call the send_auxiliary_functions_preferred_assignment every time a "new" preferred input unit is ready or if i should wait for [...]

I think this is up to you. I think both ways are allowed by the standard, just make sure to call it sometime (even if no preferred input is detected) as only after it has been received by the VT, then the operator can make assignments himself. 

Also; make sure to monitor the preferred assignment response from the VT. That should tell you useful information for finding out what's wrong:
https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/80e95fe68e61ce9b6f33670cca2f700ed53e5d43/isobus/src/isobus_virtual_terminal_client.cpp#L2590-L2623

Matthieu-Poulain, 2025-02-06:
Hi guys, just to tell you that the tests were successfull, i was able to get my assignements back after a reboot of the esp32 or the VT. Some cases though are still a bit blurry to my, i'll try to analyze what happened and get back to you later.

 I'm not sure if it will be fully understandable without the code part, but if you are interrested, here is the esp32 Log. I'll share the code modifications with you when i'll get some time next week or week end.

```ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x510
load:0x4086c110,len:0xc58
load:0x4086e610,len:0x2a30
entry 0x4086c110
I (145) cpu_start: Unicore app
I (154) cpu_start: Pro cpu start user code
I (154) cpu_start: cpu freq: 160000000 Hz
I (154) app_init: Application information:
I (157) app_init: Project name:     ISOBUS_C6_VSC_BLOB
I (163) app_init: App version:      1
I (167) app_init: Compile time:     Dec 18 2024 14:40:08
I (173) app_init: ELF file SHA256:  1620c615f...
I (178) app_init: ESP-IDF:          v5.3.1
I (183) efuse_init: Min chip rev:     v0.0
I (188) efuse_init: Max chip rev:     v0.99 
I (193) efuse_init: Chip rev:         v0.0
I (198) heap_init: Initializing. RAM available for dynamic allocation:
I (205) heap_init: At 40810A20 len 0006BBF0 (430 KiB): RAM
I (211) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (217) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (224) spi_flash: detected chip: generic
I (228) spi_flash: flash io: dio
W (232) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (248) sleep: Configure to isolate all GPIO pins in sleep state
I (252) sleep: Enable automatic switching of GPIO sleep configuration
I (259) coexist: coex firmware version: 4482466
I (274) coexist: coexist rom version 5b8dcfa
I (274) main_task: Started on CPU0
I (274) main_task: Calling app_main()
I (283) NVS_HQ: NVS: INIT SUCCESS
File Mode selected: 0
Selected IOP : CPRA_200x200
I (286) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (295) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (305) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (314) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
IOP Size : 47821
E (2325) HQ_PROCESS: INIT ISOBUS HQ 200x200  V1.0.0 : DONE;

X_Pixels: 0
Y_Pixels: 0;
VT_VERSION: 5
***  NOTE  ***
0 = Version2OrOlder
1 = Version3
2 = Version4
3 = Version5
4 = Version6
5 = ReservedOrUnknown
***  !NOTE  ***
I (24408) AgIsoStack: [Info][NM]: Partnered control function with name a0001d000cc00067 has claimed address 130 on channel 0.
I (25059) AgIsoStack: [Info][VT]: VT Server has a matching label for V1.7   . It will be loaded and upload will be skipped.
I (25105) AgIsoStack: [Info][VT]: Loaded object pool version from VT non-volatile memory with no errors.
I (25106) AgIsoStack: [Info][HQ_DBG]: LOADING PREF ASSIGNMENTS (Step 0).
LOADING DATA BLOB RESULT :0             DONE!
Device name: 0xa01a39000cc00067
Model ID: 0x55
	Number of functions: 2   
	        Function type: 2
                inputObject ID: 32013
                functionObjectID: 31000
[... 767 lines cut ...]
                inputObject ID: 32011
                functionObjectID: 31002
                        function assigned
E (129492) AgIsoStack: [Error][VT]: The VT Server is NACK-ing our VT messages. Disconnecting.
AUX_INPUT CLEAR HEREVT DISCONNECTED - Clearing assignedAuxiliaryInputDevices
I (129542) AgIsoStack: [Info][AUX-N]: New auxiliary input device with name: 11536596067839639655 and model identification code: 85
I (129544) AgIsoStack: [Info][HQ_DBG]: LOADING PREF ASSIGNMENTS (Step 3).
LOADING DATA BLOB RESULT :0             DONE!
Device name: 0xa01a39000cc00067
Model ID: 0x55
	Number of functions: 2            
		Function type: 2
                inputObject ID: 32013
                functionObjectID: 31000
                        function assigned
                Function type: 2
                inputObject ID: 32011
                functionObjectID: 31002
                        function assigned
I (129582) AgIsoStack: [Info][NM]: Control function with address 130 and NAME a0001d000cc00067 is now offline on channel 0.
I (129592) AgIsoStack: [Info][NM]: Control function with address 200 and NAME a01a39000cc00067 is now offline on channel 0.
I (129604) AgIsoStack: [Info][NM]: Control function with address 236 and NAME a0008b000cc00123 is now offline on channel 0.
I (129617) AgIsoStack: [Info][NM]: Control function with address 240 and NAME a00086000cc006d4 is now offline on channel 0.
I (129630) AgIsoStack: [Info][NM]: Control function with address 247 and NAME a00082000cc00000 is now offline on channel 0.
W (129642) AgIsoStack: [Warn][TP]: Closing active session as the destination control function is no longer valid```

Matthieu-Poulain, 2025-02-17:
Hi @GwnDaan & @ad3154,

Hope you guys are doing great.
So i tried to clean my code and merged it with the latest commit, however i am an absolute noob regarding git use so i'm not sure what i should do to provide it to you. I have a .rar file with both original commit and one with my modifications, should i send it to you?

[AgIsoStack-plus-plus-main_17_02_2025.zip](https://github.com/user-attachments/files/18826309/AgIsoStack-plus-plus-main_17_02_2025.zip)

Keep in mind that i made it so that it works with my esp32, however the file engine that was done to ensure compatibility with all plateforms is not implemented  (i.e. I bypassed it).

I also added some variables that would maybe need validation regarding their place & name used.

gunicsba, 2026-11-05:
Linking the AUX-N work that has landed since: #655 added preferred assignments to the VT client, and #763 fixes protocol differences between VT versions, which fits functions showing up on one tractor's VT but not another's. Once #763 is merged, could you retest on `main` with the tractor where the functions didn't appear? If the multi-pool approach from @cerdei solved it for you in the meantime, let us know and we'll close this.