Open-Agriculture/AgIsoStack-plus-plus #522: Faulty Object Parent 65535 Pool error bitmask value 1

Matchap1, 2024-12-30:
Hi, hope you can help. I'm quite new to this and finding your platform very interesting and helpful.

I'm using a ESP32 and trying to upload the VT test pool (VT3). I'm getting an error after uploading the object pool to the VT. Below is the output from my serial monitor: -

I (554) AgIsoStack: [Info][NM]: Partnered control function with name a0001d100cdf412c has claimed address 38 on channel 0.
I (1764) AgIsoStack: [Info][VT]: No version label from the VT matched. Client will upload the pool and store it instead.
E (47784) AgIsoStack: [Error][VT]: Error in end of object pool message.Faulty Object 65535 Faulty Object Parent 65535 Pool error bitmask value 1

The VT is a MF Datatronic 5. It doesn't seem to like the "65535" from the object pool. When I look at the object pools in the library I notice that they all start with this and it is labelled "UNDEFINED". I assume I'm missing something and after spending a few days trying to get to the bottom of the issue I though I'd reach out and see if you can point me in the right direction?

Many thanks

ad3154, 2024-12-30:
The faulty object IDs only matter if they are not 65535. 65535 is the null object ID. There isn't actually an object with that ID in any object pool.

That `error bitmask value 1` means `method or attribute not supported by the VT` which means something in that pool is unsupported by your VT, but it's not a lot of information to go on without also getting a CAN trace.

It could be something like that VT only supports 16 (or even 2) colors, but our example pool uses 8 bit color.

Matchap1, 2024-12-30:
Hi, I've found the issue.

When having trouble building my application I had removed the "-1" from the below line in the code: -

virtualTerminalClient->set_object_pool(0, testPool, (object_pool_end - object_pool_start)-1, "ais1");

After putting it back in it worked perfectly and I finally got it working on my VT. On to the Seeder example next.

I'm just writing this as I've seen your response. Thank you for your help and very fast response.

Regards Matthew

closed 2024-12-30 by Matchap1

ad3154, 2024-12-30:
Ah, nice, OK! Glad you were able to figure it out!