Open-Agriculture/AgIsoStack-plus-plus #561: Target address of the Diagnostic protocol messages

martonmiklos, 2025-03-26:
**Is your feature request related to a problem? Please describe.**
I have implemented handling of the ProductIdentification Diagnostic protocol message in AgIsoVT:
https://github.com/martonmiklos/AgIsoVirtualTerminal/blob/add_diag_protocol/src/
ServerMainComponent.cpp#L57

But the implement does not display these information properly.

I have compared the behavior of our implementation (disregard the Wireshark Transport protocol not initialized warnings):
[image]

To a physical Kverneland Tellus Pro:

[image]

The Kverneland terminal sends the response properly to the source address, while the AgIsoStack++ sends it to broadcast. 

According to the SAE J1939-21:

[image]

**Describe the solution you'd like**
Send the responses to the Diag protocol PGN request to the requestor address.

ad3154, 2025-06-03:
If I had to guess why this is happening, it's probably because the PGN is a global PGN (0xFC8D), so sending it via TP.CM to a specific destination is odd/unusual

ad3154, 2025-07-16:
I did some investigating and indeed the standard dictates that it should not be sent as a broadcast in this situation. Confirmed as a bug.