Open-Agriculture/AgIsoStack-plus-plus #585: Memory leak from create_internal_control_function on exit

bjorn-jaes, 2025-06-13:
*Note I fixed the leak by passing the internal control function to deactivate_control_function, just not sure if its intended behavior as its never done in the examples, nor mentioned in the documentation.*

**Describe the bug**
When exiting my application the leak sanitizer reports that the internal control functions creates a leak.

```c++
auto ecu isobus::CANNetworkManager::CANNetwork.create_internal_control_function(
      isobus::NAME(name), bus_id, application_settings_.hardware.can.preferred_address);
```

**Supporting Documentation**
I have included the leak sanitizer log [LeakSanitizerInternalController.txt](https://github.com/user-attachments/files/20727819/LeakSanitizerInternalController.txt)  (First glib leak can be ignored), and my [application.txt](https://github.com/user-attachments/files/20727846/application.txt) (txt as cpp was not allowed) where the leaked code was called from. (Thought I dont imagine its all that useful.)

**Environment**

- OS: 20.04
- Compiler: GCC 9.4.0
- CAN Driver: Socket CAN

**Additional context**
When deactivating the internal control function before exiting no memory leaks were reported (other than glib).

```c++
isobus::CANNetworkManager::CANNetwork.deactivate_control_function(ecu);
```
**Question**

Is this intended behavior? As the tutorial and examples dont show deactivate_control_function being used, nor does the documentation for create_internal_control_function mention that the Internal control function needs to be deactivated before exiting.

mitchelhubert, 2025-06-20:
We had a problem of communication lost with the virtual terminal that could not be restored.
Even when we de-scoped all objects and reinitialized everything it still would not restore the communication.
Only restarting the program would restore the communication.
But after reading this, we added deactivate_control_function to the end of terminating everthing.
Then when we re-initialize everthing it restores the commication, no restart needed.

So apart from a memory leak, this bug also prevents a true restart.

gunicsba, 2026-11-05:
Hello @mitchelhubert  and @bjorn-jaes  could you retest with the latest stack version?

bjorn-jaes, 2026-11-12:
I installed the latest main branch as of writing. Then commented out the deactivate code code and fixed some minor isobus namespace changes.
`isobus::CANNetworkManager::CANNetwork.deactivate_control_function(ecu);`

I now no longer get the memory leak in the sanitizer when I run my application. So I would assume that the issue has been resolved.

closed 2026-11-12 by bjorn-jaes