Open-Agriculture/AgIsoStack-plus-plus #598: Data Distionary Value formatting issue.

Rickscott23, 2025-08-05:
I've noticed there is an issue regarding the format_value method https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/main/isobus/src/isobus_data_dictionary.cpp#L41. Using a Muller Maxi ecu, all the values such as Actual Working Width (DDI 67) give the value directly, not using the offset which seems to comply with how isobus.net reports the value range to be 0 to 2147483647. Maybe I'm not understanding where the offset is used correctly but reading the raw value, it shows as the correct number and gets the offset incorrectly added to it.

I see there is an option to disable the dictionary but is there some flag I'm missing which dictates which values should be offset?

GwnDaan, 2025-08-06:
Hi @Rickscott23 

Thanks for reporting your issue.
Are you able to give us more insights into the numbers you are seeing? I'd like to see with which exact amount the raw number differs from the formatted one. That would be very much appreciated

Rickscott23, 2025-08-06:
Hi @GwnDaan 

Yes, the dictionary looks correct. here are a few examples

[Wed Aug  6 22:04:29 2025] DEBUG: Received Implement Master Switch State (DDI 141, Element 2) -> ON
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Work State with value On OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Actual Volume Per Area Application Rate as [mm³/m²] (DDI 2, Element 2) -> 0
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Volume Per Area Application Rate as [mm³/m²] with value 21474836mm³/m² OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Application Total Volume as [L] (DDI 80, Element 2) -> 0
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Application Total Volume as [L] with value 2147483648L OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Total Area (DDI 116, Element 2) -> 0
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Total Area with value 2147483648m² OK.
[Wed Aug  6 22:04:29 2025] DEBUG: TC: Received unhandled on_value_command for DDI 265, Element 0, Value 0
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 0 DDI Remaining Area with value 2147483648m² OK.
[Wed Aug  6 22:04:29 2025] DEBUG: TC: Received unhandled on_value_command for DDI 158, Element 2, Value 1
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Prescription Control State with value On OK.
[Wed Aug  6 22:04:29 2025] DEBUG: TC: Received unhandled on_value_command for DDI 308, Element 2, Value 1000000
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Percentage Application Rate with value 2148483648ppm OK.
[Wed Aug  6 22:04:29 2025] DEBUG: TC: Received unhandled on_value_command for DDI 37, Element 2, Value 0
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Volume Per Time Application Rate with value 2147483648mm³/s OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Actual Volume Content (DDI 72, Element 3) -> 14000
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 3 DDI Actual Volume Content with value 2147497648ml OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Implement Auto-Control State (DDI 160, Element 2) -> AUTO
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Section Control State with value On OK.
[Wed Aug  6 22:04:29 2025] DEBUG: TC: Received unhandled on_value_command for DDI 67, Element 2, Value 30000
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Working Width with value 2147513648mm OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Decoded Implement Section States: S1=1 S2=1 S3=1 S4=1 S5=1 S6=1 S7=1 S8=1 S9=1 S10=1 
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Condensed Work State (1-16) with value 1 1 1 1 1 1 1 1 1 1 3 3 3 3 3 3 OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Implement Master Switch State (DDI 141, Element 0) -> OFF
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 0 DDI Actual Work State with value Off OK.
[Wed Aug  6 22:04:29 2025] DEBUG: Received Implement Master Switch State (DDI 141, Element 2) -> ON
[Wed Aug  6 22:04:29 2025] DEBUG: [ISOBUS] [TC Server]: Client 135 value command for element 2 DDI Actual Work State with value On OK.

so when i read the value directly without passing through the dictionary, the values are correct but then when the library passes them and applies the offsets it seems to be off by the amount expected.

the values in the DDOP are the actual (not offset) values. for this particular Muller ECU it doesnt seem to use any offsetting but I'm not sure how we would be able to tell the difference on whether to offset or not.

closed 2025-11-04 by GwnDaan