Open-Agriculture/AgIsoStack-plus-plus #601: user interface / VT simulator

juanmartin84, 2025-08-30:
Hello, i'm starting a project.
Can somebody give some guide or point where is the info related to the creation of the user interface?
Then, is there some VT simulator to test the code?
Br
Juan

martonmiklos, 2025-08-31:
> Thanks [@martonmiklos](https://github.com/martonmiklos), is this USB-CAN interface ok for this VT simulator? https://www.inno-maker.com/product/usb2can-cable/ Regards

On Linux everything which is supported by socketcan should work.
The mentioned device looks to be supported by the stack, but I do not have on hand experience:
https://github.com/Open-Agriculture/AgIsoStack-plus-plus/blob/main/hardware_integration/src/innomaker_usb2can_windows_plugin.cpp

ad3154, 2025-09-26:
@juanmartin84  Just following up on this - did you get all the info you need, or can we clarify anything else before closing this out?